cyanfish/naps2 #158: Preview changes Brightness, Contrast, Black & White are not applied to PNG saved file

JosepBalague, 2023-06-13:
**Describe the bug**
When I change the Brightness like Contrast like Black & White the changes are not applied to the final PNG,

**To Reproduce**
Steps to reproduce the behavior:
1. Create Profile Settings (B=0, C=0)
2. Scan any page
3. Double click in preview
4. Set B&W to 1000 and press OK (Brightness and Contrast fails too)
5. Click X at preview Window
6. Save as PNG.
7. Open PNG in Paint.net: result is exactly the same when you do not run 3 to 5 steps

**Expected behavior**
Changes made in the preview should be applied to the final PNG.

**Screenshots**
[image]

**Desktop (please complete the following information):**
 - Microsoft Windows [Version 10.0.22621.1778]
 - Naps2 [7.0b8]
 - Paint.net 5.0.6

**Additional context**
n/a

cyanfish, 2023-06-13:
This does look like a bug when scanning in Black & White. Though I'm not sure the point of changing brightness/contrast etc when the image is already fully Black & White :)

cyanfish, 2023-06-13:
Gotcha, sounds like what you want is to scan in Grayscale then (not Black and White).

JosepBalague, 2023-06-13:
> Gotcha, sounds like what you want is to scan in Grayscale then (not Black and White).

I never use gray scale nor color: only B&W

cyanfish, 2023-06-13:
What happens when you try scanning in grayscale?

JosepBalague, 2023-06-13:
Wow! much better. Pale images much better in Gray scale and B&W increased in preview.

[image]

Thank you so much. Unless you tell me otherwise, I'm closing the ticket.

NextTherapist, 2023-06-14:
> This does look like a bug when scanning in Black & White. Though I'm not sure the point of changing brightness/contrast etc when the image is already fully Black & White :)

Yes, I could confirm it for scanning B&W PDFs. Tuning brightness and contrast may be difficult in dualtone images, but there should be some effect, since tuning them manually in NAPS also makes a difference...

(I would like to scan a little bit brighter to avoid visible artefacts from paper defects etc.)

cyanfish, 2023-06-18:
Black & white appearing to work in the preview window is actually a bug - it only appears to work as the preview window scales the image down which blends nearby pixels together.

For now the only way to make it work properly is to scan in grayscale first. But I have filed #159 as some scanners do have a "threshold" option that NAPS2 currently doesn't use.

closed 2023-06-18 by cyanfish