cyanfish/naps2 #160: Imported .jpg images are reencoded => quality loss

JPDelprat, 2023-06-19:
Hello, I'm not sure this is really an issue, but this is something that prevents me from using NAP2.

When I import a .jpeg file, it is re-encoded with a default quality instead of being taken as is. So, as my input .jpeg file is high quality, I get a big quality loss.

This didn't happen in 6.1.2 version.

See ImageImporter/Import => scanningContext.CreateProcessedImage(...quality=-1)

When imported file is already in a supported format, would it be possible to just copy it to recovery directory, without re-encoding.

Thanks for your answer.

cyanfish, 2023-07-09:
There should now be one less encode in [7.0b9](https://www.naps2.com/download).

JPDelprat, 2023-07-09:
Thank you very much.

JPDelprat, 2023-07-09:
Thanks very much.

I see from the modified code that the imported image is  copied to the repository instead of being re-encoded, so there is no encoding at this stage.

 

However, I’ve tested the new  version of my test case :

*	Input 22 high-quality JPEG (encoded with very few compression) for a total size of 63.4 MB
*	Resulting pdf with ersion 7.0b9 is only 13 MB
*	With 6.1.2 version, pdf size was 63.4 MB (same size than input ; jpeg files seem to be re-encoded, but with the same quality).

 

Ideally, I would like JPEG to be inserted in pdf files without any encoding, so that I can extract them losslessly.

 

Maybe I’m the only person to expect this behavior. It is pretty specific. I try to keep the maximum quality from scan to pdf. I even appreciate to be able extract images from pdf and do some changes if needed, « without  loss ».

 

 

Thanks

cyanfish, 2023-09-05:
JPEGs should now be embedded directly in PDFs in [7.1.0](https://www.naps2.com/download). This also leads to a big speedup for PDF saving in a lot of cases.

JPDelprat, 2023-09-06:
Thanks very much. This is exactly what I was hoping.

JPDelprat, 2023-09-06:
Hello, I'm sorry but I've justed the new version and I still have the same issue (pdf is about half the size of my images). Do I have to activate some option ?
Thanks

cyanfish, 2023-09-06:
The issue seems to be that the JPEG you have uses an EXIF header while NAPS2 relies on a JFIF header. I'll see if I can add EXIF support at some point but it should be possible to convert EXIF JPEGs to JFIF JPEGs if you want.

JPDelprat, 2023-09-06:
Thanks for the analysis.
Too bad : all my JPEG seem to have an EXIF header...

I had a look to your code changes, and suspected this part of the code too. 
I didn't know there were several kinds of JPEG headers. 
Do you need to parse JPEG header only to determine if the image is grayscaled, or are there other reasons ? 
Maybe you could you a standard image reading library, such as ImageMagick or any other one.
I will see if I can write some code to convert my JPEGs from EXIT to JFIF losslessly.
Thanks.

cyanfish, 2023-09-06:
Doing my own header parsing isn't strictly required but it's 100x faster than actually loading the image using the image library I have available.

The JFIF/EXIF part of the header is mainly needed for reading the image resolution/dpi, which is used to determine the physical size of the generated PDF. For some people that's not relevant but other times it can be the difference between things printing properly vs not.

JPDelprat, 2023-09-10:
It looks like you've already made the changes. Thank you. I look forward to the new version.

cyanfish, 2023-10-11:
This should be fixed in [7.1.1](https://www.naps2.com/download).

closed 2023-10-11 by cyanfish

JPDelprat, 2023-10-14:
It now works for my use case ! Thanks a lot.

commit 31cd1bbf (2023-07-09): Avoid re-encoding imported jpegs for storage

Fixes #160