cyanfish/naps2 #162: Replace and Insert page

trteder, 2023-06-20:
**Is your feature request related to a problem? Please describe.**
I'm installing NAPS2 in a high production environment and I'm missing some features:
1 - Insert missing page. 
2 - Replace page.
3 - Insert batch number.

**Describe the solution you'd like**
1 - Insert missing page. When selecting a certain page and scanning, the scanned page will be immediately added to the selection, without having to move to the point after scanning.

2 - Replace page. When noticing that a certain page did not have the expected quality (crooked, misaligned, folded), the operator clicks on the page(s) to be replaced and clicks on scan, exchanging the current pages for new pages.

3 - Insert batch number. When clicking on batch scan, allow the batch number to be typed, which can be used in the name of the file(s) generated, for example, using the letter L in the substitution (llllll).

**Describe alternatives you've considered**
We are looking to components or sdk to solve these problems, but without success. 

**Additional context**
I understand that these features will greatly enrich the software. Thank you very much in advance.

organist1216, 2023-08-21:
As someone who primarily uses NAPS in tandem with Scantailor, I would like to have the ability to insert missing pages and replace existing pages as mentioned above, but with the ability to select a file from my computer rather than being limited to scanning a new page.

trteder, 2023-08-22:
@cyanfish, I made these changes in my repository, obviously not using good practices and/or following project standards, since I don't know much about it, but I understand that it can be used as a basis for understanding. Thank you very much.