cyanfish/naps2 #165: Launch abort on macOS (M1 Ventura)

giabao06, 2023-06-27:
**Describe the bug**
I downloaded the latest version as of writing this (7.0b8) for macOS (Apple Silicon) and attempted to launch it and it crashed (`SIGABRT`), even after updating from Monterey to Ventura

**To Reproduce**
Steps to reproduce the behavior:
1. Download the latest NAPS2 from naps2.com
2. Install the app
3. Run it

**Expected behavior**
App would start normally

**Screenshots**
(N/A)

**Desktop (please complete the following information):**
 - OS: macOS 13
 - Version 13.4.1

**Additional context**
Tried both the M1 and Intel versions as well as older versions, same result. Running from terminal throws this error:

```
2023-06-27 12:36:28.693 NAPS2[3429:44324] Could not find SixLabors.ImageSharp, Version=1.0.0.0, Culture=neutral, PublicKeyToken=d998eea7b14cab13 referenced by assembly PdfSharpCore, Version=1.0.0.0, Culture=neutral, PublicKeyToken=null.
Unhandled exception. System.Reflection.TargetInvocationException: Arg_TargetInvocationException
 ---> System.UnauthorizedAccessException: UnauthorizedAccess_IODenied_Path, /Users/tranpgb/.config/naps2/recovery
 ---> System.IO.IOException: Permission denied
   Exception_EndOfInnerExceptionStack
   at System.IO.FileSystem.CreateDirectory(String fullPath, UnixFileMode unixCreateMode)
   at System.IO.Directory.CreateDirectory(String path)
   at NAPS2.Paths.EnsureFolderExists(String folderPath)
   at NAPS2.Paths.get_Recovery()
   at NAPS2.Modules.RecoveryModule.Load(ContainerBuilder builder)
   at Autofac.Module.Configure(IComponentRegistryBuilder componentRegistry)
   at Autofac.Core.Registration.ModuleRegistrar.<.ctor>b__1_0(IComponentRegistryBuilder reg)
   at Autofac.ContainerBuilder.Build(IComponentRegistryBuilder componentRegistry, Boolean excludeDefaultModules)
   at Autofac.ContainerBuilder.Build(ContainerBuildOptions options)
   at NAPS2.Modules.AutoFacHelper.FromModules(IModule[] modules)
   at NAPS2.EntryPoints.MacEntryPoint.Run(String[] args)
   at System.RuntimeMethodHandle.InvokeMethod(Object target, Void** arguments, Signature sig, Boolean isConstructor)
   at System.Reflection.MethodInvoker.Invoke(Object obj, IntPtr* args, BindingFlags invokeAttr)
   Exception_EndOfInnerExceptionStack
   at System.Reflection.MethodInvoker.Invoke(Object obj, IntPtr* args, BindingFlags invokeAttr)
   at System.Reflection.RuntimeMethodInfo.Invoke(Object obj, BindingFlags invokeAttr, Binder binder, Object[] parameters, CultureInfo culture)
   at NAPS2.Program.Main(String[] args)
zsh: abort      ./[NAPS2.app/Contents/MacOS/NAPS2](http://naps2.app/Contents/MacOS/NAPS2)
```

cyanfish, 2023-06-27:
That's weird, it looks like somehow your filesystem permissions got messed up or some kind of corruption happened. I would recommend opening a terminal and running:
`rm -r /Users/tranpgb/.config/naps2`
or if that doesn't work
`sudo rm -r /Users/tranpgb/.config/naps2`

cyanfish, 2023-06-27:
Actually it seems like the issue is that another application may have created the `/Users/tranpgb/.config` folder with the wrong permissions.

You might be able to fix it by running this:
`chmod 755 /Users/tranpgb/.config`

cyanfish, 2023-06-27:
If that doesn't work please post the output of `ls -la /Users/tranpgb/.config`.

giabao06, 2023-06-27:
Dosen't work, still throws the same error. 

Output of `ls -la /Users/tranpgb/.config`: 

```
tranpgb@[censored]/Applications % ls -la /Users/tranpgb/.config
total 0
drwxr-xr-x   3 root     staff    96 Sep  8  2022 .
drwxr-x---+ 33 tranpgb  staff  1056 Jun 27 12:53 ..
drwx------   3 root     staff    96 Sep  8  2022 htop
```

cyanfish, 2023-06-27:
Ok looks like the owner is wrong, I think `sudo chown tranpgb /Users/tranpgb/.config` should work

giabao06, 2023-06-27:
Working now, thank you. Closing issue.

closed 2023-06-27 by giabao06