cyanfish/naps2 #166: NAPS2 7.0beta8 failing to open Outlook 2013 to send mail

SelfMan, 2023-06-28:
When trying to use the "Email PDF", NAPS is unable to open the new mail window and shows the error:

System.Exception: Could not load dll for email: C:\Program Files\Microsoft Office 15\root\VFS\ProgramFilesCommonX86\system\msmapi\1051\msmapi32.dll
   at ValueTuple<MapiSendMailDelegate, MapiSendMailDelegateW> NAPS2.ImportExport.Email.Mapi.SystemEmailClients.GetDelegate(string clientName, out bool unicode)
   at Task<MapiSendMailReturnCode> NAPS2.ImportExport.Email.Mapi.MapiWrapper.SendEmail(string clientName, EmailMessage message)+() => { }
   at async Task<MapiSendMailReturnCode> NAPS2.ImportExport.Email.Mapi.MapiDispatcher.SendEmail(string clientName, EmailMessage message)
   at async Task<bool> NAPS2.ImportExport.Email.Mapi.MapiEmailProvider.SendEmail(EmailMessage message, ProgressHandler progress)+(?) => { }
   at async bool NAPS2.Pdf.SavePdfOperation.Start(string fileName, Placeholders placeholders, ICollection<ProcessedImage> images, PdfSettings pdfSettings, OcrParams ocrParams, EmailMessage emailMessage, string overwriteFile)+(?) => { } [1]

The 6.1.2 release works just fine and works as expected.

OS: Windows 10 Pro 22H2 x64 with all updates installed (10.0.19045.3086)
Office 2013 x86 (15.0.5553.1000)

I've also tested this with Outlook 2019 and it does have the same problem.

SelfMan, 2023-06-28:
System.Exception: Could not load dll for email: C:\Program Files (x86)\Microsoft Office\root\VFS\ProgramFilesCommonX86\system\msmapi\1051\msmapi32.dll
   at ValueTuple<MapiSendMailDelegate, MapiSendMailDelegateW> NAPS2.ImportExport.Email.Mapi.SystemEmailClients.GetDelegate(string clientName, out bool unicode)
   at Task<MapiSendMailReturnCode> NAPS2.ImportExport.Email.Mapi.MapiWrapper.SendEmail(string clientName, EmailMessage message)+() => { }
   at async Task<MapiSendMailReturnCode> NAPS2.ImportExport.Email.Mapi.MapiDispatcher.SendEmail(string clientName, EmailMessage message)
   at async Task<bool> NAPS2.ImportExport.Email.Mapi.MapiEmailProvider.SendEmail(EmailMessage message, ProgressHandler progress)+(?) => { }
   at async bool NAPS2.Pdf.SavePdfOperation.Start(string fileName, Placeholders placeholders, ICollection<ProcessedImage> images, PdfSettings pdfSettings, OcrParams ocrParams, EmailMessage emailMessage, string overwriteFile)+(?) => { } [1]

closed 2023-07-09 by cyanfish

cyanfish, 2023-07-09:
Please try with [7.0b9](https://www.naps2.com/download).

SelfMan, 2023-07-09:
Hello, thank you for the fix.
I am confirming that the issue was resolved in NAPS2 7.0 beta 9

commit 900b9042 (2023-07-09): Fix mapi call

Fixes #166