cyanfish/naps2 #167: How to compile Linux arm64 version,NO flatpak

kero990, 2023-07-04:
I tried compiling a non flatpak naps2 for my arm64 platform because I saw that the official release includes Linux arm64

But I can only compile x64 binary files using the official compilation process, even if I compile on the arm platform and use arm64's dotnet CLI

May I ask if there is a way to compile an arm64 Linux program? Sorry, I really don't need Flatpak

cyanfish, 2023-07-04:
I've not set up linux-arm64 builds yet, but if you care to, you can try modifying the [packaging code](https://github.com/cyanfish/naps2/blob/45c307af3842f820352fe73a30bc55a606932a78/NAPS2.Tools/Project/Targets/TargetsHelper.cs#L119) to use `Platform.LinuxArm` instead of `Platform.Linux`, then [run](https://github.com/cyanfish/naps2/wiki/2.-NAPS2.Tools) `n2 pkg deb` (or rpm).

Note that OCR won't work as that relies on a custom tesseract build that is also not built for linux-arm64 yet. I think everything else should work but I'm not sure as I haven't tried it.

kero990, 2023-07-06:
@cyanfish

Yes, I compiled successfully, it seems that all other functions are normal.

But I'm not satisfied yet. I want to replace tesseract to fix the ORC feature, as tesseract itself should be able to compile and run under arm64.

I see in your project library that there are

https://github.com/cyanfish/naps2-tesseract

But I don't know how to compile this project, nor do I know how to replace it in naps2. Does it also need to be compiled using the n2 tool?

cyanfish, 2023-07-06:
That one's a lot more of a hassle... I'll try and update that myself at some point in the next couple weeks.

But if you really want to, the steps would be something like this:
- Make sure to load git submodules (into ./sources)
- There are a couple patches in ./patches for the submodules to apply - you can probably skip this step if you don't care about the binary size
- The build scripts are in ./scripts - you can copy the x64 one and change the paths, it should be fairly simple as you're not cross-compiling
- That should generate binaries in ./artifacts
- You'll need to update the .csproj and .targets [files](https://github.com/cyanfish/naps2-tesseract/tree/master/NAPS2.Tesseract.Binaries) to reference/copy the new artifacts (with "_linuxarm" instead of "_linux")
- You'll need to bump the .csproj version number and build it to generate a .nupkg
- You'll need to set up a [local package source](https://learn.microsoft.com/en-us/nuget/hosting-packages/local-feeds) with that package to use when building NAPS2
- Then bump the version of the NAPS2.Tesseract references in NAPS2 to match the new version number

cyanfish, 2023-07-06:
I've updated naps2-tesseract with a linux-arm64 build so you can disregard the above. All you need to do is change the version reference for NAPS2.Tesseract.Binaries from 1.0.5 to 1.1.0 in the NAPS2 csproj files.

cyanfish, 2023-07-09:
I've also now added official arm64 deb/rpm builds in [7.0b9](https://www.naps2.com/download).

kero990, 2023-07-09:
I haven't tested the ocr function yet, but I found that the current import function is damaged. After importing the PDF, an error is reported as follows

```
System.Exception: Could not load library: "/usr/lib/naps2/_linuxarm/libpdfium.so". Error: 
   at NAPS2.Unmanaged.NativeLibrary.DoLoadLibrary(String path)
   at System.Lazy`1.ViaFactory(LazyThreadSafetyMode )
   at System.Lazy`1.ExecutionAndPublication(LazyHelper , Boolean )
   at System.Lazy`1.CreateValue()
   at System.Lazy`1.get_Value()
   at NAPS2.Unmanaged.NativeLibrary.get_LibraryHandle()
   at NAPS2.Unmanaged.NativeLibrary.LoadFunc[T]()
   at NAPS2.Unmanaged.NativeLibrary.<Load>b__12_0[T]()
   at NAPS2.Util.CollectionExtensions.Get[TKey,TValue](Dictionary`2 dict, TKey key, Func`1 defaultValue)
   at NAPS2.Unmanaged.NativeLibrary.Load[T]()
   at NAPS2.Pdf.Pdfium.PdfiumNativeLibrary.get_FPDF_InitLibrary()
   at NAPS2.Pdf.Pdfium.PdfiumNativeLibrary.<>c.<.cctor>b__201_0()
   at System.Lazy`1.ViaFactory(LazyThreadSafetyMode )
   at System.Lazy`1.ExecutionAndPublication(LazyHelper , Boolean )
   at System.Lazy`1.CreateValue()
   at NAPS2.Pdf.PdfImporter.<>c__DisplayClass7_0.<Import>b__0(Action`1 produceImage)
   at NAPS2.Util.AsyncProducers.<>c__DisplayClass0_0`1.<RunProducer>b__0(Action`1 produce)
   at NAPS2.Util.AsyncProducers.<>c__DisplayClass1_0`1.<<RunProducer>b__0>d.MoveNext()
   at async IAsyncEnumerable<T> NAPS2.Util.AsyncSink<T>.AsAsyncEnumerable()+MoveNext()
   at async bool NAPS2.ImportExport.ImportOperation.Start(List<string> filesToImport, Action<ProcessedImage> imageCallback, ImportParams importParams)+(?) => { } x 2
```

cyanfish, 2023-07-09:
I've updated the arm64 builds with a fix, can you try redownloading/installing?

kero990, 2023-07-10:
The import function seems to be working properly, but there are still errors in the button bar.

I can drag and drop a PDF file to import into Naps2, but clicking the import button has no response and cannot pop up a file selection box. Additionally, when I click on Print, the program will flash back directly.

I confirm that the same issue existed in the previous version. I cleared the configuration file of~/. config/naps2, and the old version did indeed have this issue. But I'm not sure whether it has anything to do with my Desktop environment. I use DDE desktop, which is not very common

I have captured the following records in the errorlog.txt file, which should include errors when I click on import or export, but I am not sure if the records that crashed directly when I click on print can be saved.

```
2023-07-10 10:05:32.9527 9592 An unhandled error occurred. System.Reflection.TargetInvocationException: Exception has been thrown by the target of an invocation.
 ---> GLib.GException: 转换输入中出现无效字符序列
   at IntPtr GLib.Marshaller.StringToFilenamePtr(string str)
   at bool Gtk.FileChooserDialog.SetCurrentFolder(string filename)
   at new Eto.GtkSharp.Forms.OpenFileDialogHandler()
   at void Eto.GtkSharp.Platform.AddTo(Platform p)+() => { } [96]
   at new Eto.Widget()
   at new Eto.Forms.CommonDialog()
   at new Eto.Forms.FileDialog()
   at new Eto.Forms.OpenFileDialog()
   at bool NAPS2.EtoForms.EtoDialogHelper.PromptToImport(out string[] filePaths)
   at void NAPS2.EtoForms.Desktop.DesktopController.Import()
   at NAPS2.EtoForms.ActionCommand(Action action)+(object sender, EventArgs args) => { }
   at void Eto.Forms.Command.OnExecuted(EventArgs e)
   at void Eto.Forms.Command.Execute()
   at void NAPS2.EtoForms.Ui.GtkDesktopForm.CreateToolbarButton(Command command)+(object _, EventArgs _) => { } [0]
   --- End of inner exception stack trace ---
   at object RuntimeMethodHandle.InvokeMethod(object, in Span<object>, Signature, bool, bool)
   at object System.Reflection.RuntimeMethodInfo.Invoke(object, BindingFlags, Binder, object[], CultureInfo)
   at object Delegate.DynamicInvokeImpl(object[] args)
   at void GLib.Signal.ClosureInvokedCB(object o, ClosureInvokedArgs args)
   at void GLib.SignalClosure.MarshalCallback(IntPtr raw_closure, IntPtr return_val, uint n_param_vals, IntPtr param_values, IntPtr invocation_hint, IntPtr marshal_data)
2023-07-10 10:05:32.9890 9592 An error occurred that caused the application to close. System.Reflection.TargetInvocationException: Exception has been thrown by the target of an invocation.
 ---> GLib.GException: 转换输入中出现无效字符序列
   at IntPtr GLib.Marshaller.StringToFilenamePtr(string str)
   at bool Gtk.FileChooserDialog.SetCurrentFolder(string filename)
   at new Eto.GtkSharp.Forms.OpenFileDialogHandler()
   at void Eto.GtkSharp.Platform.AddTo(Platform p)+() => { } [96]
   at new Eto.Widget()
   at new Eto.Forms.CommonDialog()
   at new Eto.Forms.FileDialog()
   at new Eto.Forms.OpenFileDialog()
   at bool NAPS2.EtoForms.EtoDialogHelper.PromptToImport(out string[] filePaths)
   at void NAPS2.EtoForms.Desktop.DesktopController.Import()
   at NAPS2.EtoForms.ActionCommand(Action action)+(object sender, EventArgs args) => { }
   at void Eto.Forms.Command.OnExecuted(EventArgs e)
   at void Eto.Forms.Command.Execute()
   at void NAPS2.EtoForms.Ui.GtkDesktopForm.CreateToolbarButton(Command command)+(object _, EventArgs _) => { } [0]
   --- End of inner exception stack trace ---
   at object RuntimeMethodHandle.InvokeMethod(object, in Span<object>, Signature, bool, bool)
   at object System.Reflection.RuntimeMethodInfo.Invoke(object, BindingFlags, Binder, object[], CultureInfo)
   at object Delegate.DynamicInvokeImpl(object[] args)
   at void GLib.Signal.ClosureInvokedCB(object o, ClosureInvokedArgs args)
   at void GLib.SignalClosure.MarshalCallback(IntPtr raw_closure, IntPtr return_val, uint n_param_vals, IntPtr param_values, IntPtr invocation_hint, IntPtr marshal_data)
2023-07-10 10:05:51.2379 9592 An unhandled error occurred. GLib.GException: 转换输入中出现无效字符序列
   at IntPtr GLib.Marshaller.StringToFilenamePtr(string str)
   at bool Gtk.FileChooserDialog.SetCurrentFolder(string filename)
   at new Eto.GtkSharp.Forms.SaveFileDialogHandler()
   at void Eto.GtkSharp.Platform.AddTo(Platform p)+() => { } [99]
   at new Eto.Widget()
   at new Eto.Forms.CommonDialog()
   at new Eto.Forms.FileDialog()
   at new Eto.Forms.SaveFileDialog()
   at bool NAPS2.EtoForms.EtoDialogHelper.PromptToSaveImage(string defaultPath, out string savePath)
   at async Task<bool> NAPS2.ImportExport.ExportController.SaveImages(ICollection<UiImage> uiImages, ISaveNotify notify)
   at async Task NAPS2.EtoForms.Desktop.DesktopController.SaveImages()
   at async NAPS2.EtoForms.ActionCommand(Action action)+(?) => { }
   at void System.Threading.Tasks.Task.ThrowAsync(Exception, SynchronizationContext)+() => { }
   at void Eto.GtkSharp.Forms.GtkSynchronizationContext.Post(SendOrPostCallback d, object state)+() => { }
   at bool GLib.Idle+IdleProxy.Handler()
```

cyanfish, 2023-07-13:
For the import/save issue the root cause is https://github.com/GtkSharp/GtkSharp/issues/393.

cyanfish, 2023-12-17:
This should be fixed in [NAPS2 7.2.1](https://www.naps2.com/download).

closed 2023-12-17 by cyanfish