cyanfish/naps2 #168: [7.0b9] Scans are black and white only

akuropka, 2023-07-10:
**Describe the bug**
Scans are in black and white only despite color selected.

**To Reproduce**
Steps to reproduce the behavior:
1. Use F2 for scanning.
2. Set to scan an image in color.
3. Do the scan.

**Expected behavior**
Picture is in color as it was before including 7.0b8.

**Actual behavior**
Despite preview (if selected) is in color the scan is actually in black and white.

**Desktop (please complete the following information):**
Microsoft Windows [Version 10.0.22631.1972]
Kyocera ECOSYS M5521cdw via TWAIN v3.0.1421 (x64)

**Additional context**
Downgraded to 7.0b8 without modifying/resetting settings and picture was in color.

**Screencast**
https://github.com/cyanfish/naps2/assets/22570574/e7880a39-3fa4-4cc8-a753-285489e59ff5

cyanfish, 2023-07-10:
I see the problem, until the fix is out you can edit your profile, select "Use predefined settings", change bit depth to "24-bit Color", then select "Use native UI" again.

akuropka, 2023-07-10:
Works, thank you. 👍

euklidoid, 2023-07-26:
@cyanfish 
same issue on Mac OS 13.4.1, brother DSmobile DS-740D with Apple Driver (SANE-Driver is not working)

do you also have a workaround in place.

let me me if you want me to provide more details.

And sorry for jumping into this thread, but since it is the same issue I thought it will be ok.

cyanfish, 2023-09-05:
This should be fixed in [7.1.0](https://www.naps2.com/download).

closed 2023-09-05 by cyanfish