cyanfish/naps2 #169: Docker image

webysther, 2023-07-21:
**Is your feature request related to a problem? Please describe.**
No

**Describe the solution you'd like**
I don't know if is possible, but [like calibre](https://hub.docker.com/r/linuxserver/calibre), with vnc maybe this project can run on browser for self-host.

This is how calibre looks like on browser:
[image]

The ideia is build the linux version and get this working, but I have same questions about it:
- Can USB printer be accessible from browser over docker? A: I know this works for USB game controller for instance and I think this work for any kind of MIDI, I think...

**Describe alternatives you've considered**
Don't find one.

**Additional context**

cyanfish, 2023-07-22:
It looks like the Calibre image uses [Guacamole](https://guacamole.apache.org/), which [does not support](https://issues.apache.org/jira/browse/GUACAMOLE-522) USB redirection. In general [WebUSB](https://developer.mozilla.org/en-US/docs/Web/API/WebUSB_API) is a thing but it seems like it requires specific support from the device so it wouldn't work for typical scanners.

What scenario do you have where you don't want to install NAPS2 locally?

Based on talking about USB redirection I'm assuming you're interested in hosting NAPS2 remotely but accessing local devices, though the opposite scenario - local NAPS2 accessing remote devices - is currently [in development](https://github.com/cyanfish/naps2/tree/master/NAPS2.Escl.Server).

webysther, 2023-07-22:
I moving lots of software to container-web-based because I want to reduce the dependency of one machine and to make possible to run inside the NAS box and save the files direct in a volume mounted in the inbox of scans, this will allow me to create workflows using n8n, github actions or any other tool to process the pdf, like [Stirling-PDF](https://github.com/Frooodle/Stirling-PDF).

Games and lots of C started moving to [wasm ](https://webassembly.org/) and this show a great potential of web to run native C/C++ code on browser. An example is [emulatorjs](https://docs.linuxserver.io/images/docker-emulatorjs) that use a compile set of emulators from libreretro on browser.

webysther, 2023-07-22:
Maybe a client/server for the future of NAPS2 will be emergy from SDK, because is possible to create a web application with same look and few thats talk with a API which translate to SDK. The problem still remains on the device comunication, [I found this experiment](https://github.com/drffej/webusb.printer).

Maybe too soon for this.

webysther, 2023-07-22:
The project is not Guacamole only, but one tools, looks [like they support](https://www.reddit.com/r/kasmweb/comments/zv51k1/general_usbpassthrough/).

cyanfish, 2023-07-22:
Yes, it will definitely be possible to build a JS API on top the SDK, see the [existing issue](https://github.com/cyanfish/naps2/issues/120). That would require having a locally installed component for it to connect to.

Straight USB access from the browser does seem like it's beyond what's currently available.

webysther, 2023-07-22:
They still wait for your implementation 😅

webysther, 2023-07-22:
BTW I love this project, hope have money in the future to sponsor you.

closed 2023-07-22 by webysther