cyanfish/naps2 #170: Translated string not appearing in GUI

tmtisfree, 2023-07-28:
**Describe the bug**
A translated string is not appearing in the GUI.

**To Reproduce**
Steps to reproduce the behavior:
Open the About window: the "Enable debug logging" string is still in English even when the GUI is in French (in my case, as I am the French translator and I know this particular string is translated).

**Expected behavior**
The "Enable debug logging" string should appear in the About window.

**Screenshots**
None provided.

**Desktop (please complete the following information):**
 - Windows 10
 - Version 7.0.9.0

**Additional context**

cyanfish, 2023-07-28:
Sorry about that, I did forget to update some strings for b9, I'll get those in for the next version.

cyanfish, 2023-09-05:
Strings should be updated in [7.1.0](https://www.naps2.com/download).

closed 2023-09-05 by cyanfish