cyanfish/naps2 #171: Error Domain=com.apple.ImageCaptureCore Code=-9933

stsgit, 2023-07-29:
Hi,

I'm using NAPS 7.0.9 on MacOS 13.4.1 (Apple M2). My Scanner is a Canon TS3400 series.

When I start scanning I get the exception Error Domain=com.apple.ImageCaptureCore Code=-9933.

It seems to be a problem with the color depth. The exception occurs while trying to scan in b/w. After changing settings to grey scanning is ok.

Using the MacOS Scan-Tool I can scan in b/w.

These are similar topics I found:
https://github.com/holtwick/pdfify/issues/48
https://stackoverflow.com/questions/21502161/imagecapturecore-error-9933-when-icscannerfunctionalunit-set-to-scan-in-black

spezifanta, 2024-07-31:
I had the same problem (found this issue by searching the error code). 

After some fiddling, I changed my USB adapter to a USB 3 cable and connected the scanner directly, and voila, it worked.

al10b, 2026-09-23:
Hi, it looks like this issue is still present. Scanning in color works fine, but scanning in black and white produces this error:

```
Error Domain=com.apple.ImageCaptureCore Code=-9933 "An error occurred during scanning." UserInfo={NSLocalizedDescription=An error occurred during scanning.}
```

The log file shows:

```
2026-09-23 10:48:30.0034 21186 net_http_message_not_success_statuscode_reason, 405, Method Not Allowed System.Net.Http.HttpRequestException: net_http_message_not_success_statuscode_reason, 405, Method Not Allowed
   at HttpResponseMessage System.Net.Http.HttpResponseMessage.EnsureSuccessStatusCode()
   at async Task<EsclJob> NAPS2.Escl.Client.EsclClient.CreateScanJob(EsclScanSettings settings)
   at async Task<EsclJob> NAPS2.Scan.Internal.Escl.EsclScanDriver.CreateScanJobAndCorrectInvalidSettings(EsclClient client, EsclScanSettings scanSettings)
   at async Task NAPS2.Scan.Internal.Escl.EsclScanDriver.Scan(ScanOptions options, CancellationToken cancelToken, IScanEvents scanEvents, Action<IMemoryImage> callback)
   at async Task NAPS2.Scan.Internal.RemoteScanController.Scan(ScanOptions options, CancellationToken cancelToken, IScanEvents scanEvents, Action<ProcessedImage, PostProcessingContext> callback)
   at async Task NAPS2.Scan.ScanController+<>c__DisplayClass15_1.<Scan>g__DoScan|7(?)+DoScan(?)
   at async IAsyncEnumerable<ProcessedImage> NAPS2.Scan.ScanController.Scan(ScanOptions options, CancellationToken cancelToken)+(?) => { }
```

Some additional details:
- macOS: Tahoe 26.6.2
- Printer: HP Color LaserJet Pro MFP M277dw
- Connection: Wi-Fi

I’m looking for an alternative to HP Easy Scan, since HP no longer supports this combination. Any help would be appreciated!