cyanfish/naps2 #178: 7.0b9 crashes consistently when importing PDF

organist1216, 2023-08-21:
**Describe the bug**
I recently updated to 7.0b9 and it crashes when I try to import a PDF. It does it across multiple files (all of which worked fine in previous versions of NAPS. There are no error messages or anything, it just abruptly closes. 

**To Reproduce**
Steps to reproduce the behavior:
1. click on "import PDF"
2. select PDF to import
3. program crashes at some point when importing

**Desktop (please complete the following information):**
 - OS: Windows 11, 22H2
 - Version 22621.2134

cyanfish, 2023-08-26:
Can you open the Windows [Event Viewer](https://www.howtogeek.com/123646/htg-explains-what-the-windows-event-viewer-is-and-how-you-can-use-it/) and see if, under Application, there are any logs for the crash?

organist1216, 2023-08-29:
Sorry for the delay... There is an error listed in Event Viewer. Here are some screenshots. 
[image]

[image]
[image]
[image]

cyanfish, 2023-08-29:
Thanks, there's one entry below that one (labeled ".NET Runtime") that should have more info if you can screenshot that as well.

organist1216, 2023-08-31:
Here it is. The highlighted bit of text in one of the screenshots was just for me to gauge how much I had to scroll to get everything in 2 screenshots. 

![Screenshot 2023-08-30 224652](https://github.c
[image]
[image]
om/cyanfish/naps2/assets/127072963/6aca4102-ce36-4d49-bc9f-1a9e83905cdf)
[image]

Let me know if there is anything else...thanks for the hard work/responsiveness.

organist1216, 2023-08-31:
Left out the last one for some reason...

[image]

cyanfish, 2023-08-31:
Thanks, that's helpful. Is there a sample PDF with the problem that you'd be able to upload?

organist1216, 2023-08-31:
I tested and it crashed with this one, though the original one that I was trying is too large to upload.

[Humility Of Heart.pdf](https://github.com/cyanfish/naps2/files/12482011/Humility.Of.Heart.pdf)

cyanfish, 2023-08-31:
Thanks, that was enough for me to track the problem. It's definitely one that can be inconsistent. I'll have a fix in the next version.

cyanfish, 2023-09-05:
This should be fixed in [7.1.0](https://www.naps2.com/download).

closed 2023-09-05 by cyanfish

organist1216, 2023-09-06:
Everything works perfectly now. Thank you very much!