cyanfish/naps2 #181: Kurdish Language

mohammadtarabishy, 2023-09-05:
please i am trying to run OCR on Kurdish Language , it give me wrong result 
i try use kurdish and kurdish(arab)
how to now the usage lang in naps2 to give me correct results 
[image]
[image]
and this some of samples i trying on it

cyanfish, 2023-09-06:
Unfortunately it seems like the latest version of Tesseract doesn't actually support Arabic Kurdish (see https://github.com/tesseract-ocr/langdata/issues/124), that shouldn't even be an option to download in NAPS2 anymore. You could either try Arabic or use the [6.1.2](https://github.com/cyanfish/naps2/releases/tag/v6.1.2) version of NAPS2 which uses an older Tesseract that does support Arabic Kurdish.

cyanfish, 2023-10-11:
In NAPS2 [7.1.1](https://www.naps2.com/download) it should be listed as Kurmanji instead of Kurdish now (as that's the only supported dialect).

closed 2023-10-11 by cyanfish