cyanfish/naps2 #182: NAPS2.Portable.exe isn't high DPI aware

Chouffy, 2023-09-05:
**Describe the bug**
When using the portable version of naps2 and starting the app through `NAPS2.Portable.exe`, the application looks blurry on monitor with DPI > 100%.

**To Reproduce**
Steps to reproduce the behavior:
1. Download the portable version (`naps2-7.1.0-win.zip`)
2. Extract the `zip`
3. Launch `NAPS2.Portable.exe`

**Expected behavior**
A high-DPI aware interface.
This can be achieve with the following workaround:
1. Right-click on `NAPS2.Portable.exe` and select Properties
2. Go to the Compability tab
3. Click on Change high DPI settings
4. In the second section, check "Override high DPI scaling behavior"
5. In "Scaling performed by:", make sure "Application is selected

**Screenshots**
High-DPI unaware: title bar is blurred
[image]

High-DPI aware: nice and crisp
[image]

**Desktop (please complete the following information):**
 - OS: Windows 11
 - Version 22H2

**Additional context**
Nope, it's a small thing :)
Thanks a lot for this great piece of software!

cyanfish, 2024-10-05:
This should be fixed in [8.0b1](https://www.naps2.com/download).

closed 2024-10-05 by cyanfish

Chouffy, 2024-10-15:
Works perfectly, thanks a lot @cyanfish !