cyanfish/naps2 #186: Console application working only in visual studio debug

shahareldad, 2023-09-07:
Hi

I wrote a console application per the example in the SDK readme and managed to get it working fully via Visual Studio debug mode.
But when I go to the output exe file of the console application and test it, the output PDF is not saved.

The NAPS2.App.Console app does work and saves the PDF and I tried to search what is the difference between them but couldn't find any difference.

Any idea why when I run the application from command line the method "ToBlockingEnumerable" returns 0 documents? (The scanner takes the pages).

Thanks for any help

using NAPS2.Images;
using NAPS2.Images.Gdi;
using NAPS2.Pdf;
using NAPS2.Scan;
using System.Collections.ObjectModel;
using System.Text.Json;

internal class Program
{
    [STAThread]
    private static async Task Main(string[] args)
    {
        await new Program().Run(args);
    }

    private async Task Run(string[] args)
    {
        var configDirectory = @"c:\clscanner\";
        var filename = @"c:\temp\Adapter_Scaned_File.pdf";
        var bitDepth = BitDepth.BlackAndWhite;
        var dpi = 200;
        var deviceName = "";
        var logFile = @"c:\temp\scan.log";

        if (args.Length > 0)
        {
            if (args[0].StartsWith("ladpcscanapp:"))
            {
                args = args[0].Replace("ladpcscanapp:", "").Split(new[] { "&" }, StringSplitOptions.None);
            }
            filename = args[0];
            if (args.Length > 4) configDirectory = args[4];
            //File.AppendAllText(logFile, configDirectory + "configuration.json" + Environment.NewLine);
            var content = File.ReadAllText(configDirectory + "configuration.json");
            var configuration = JsonSerializer.Deserialize<LadpcScanAppConfiguration>(content);
            if (args.Length > 1) bitDepth = args[1] == "true" ? BitDepth.Color : configuration != null && configuration.GrayInsteadOfBW.Equals("true") ? BitDepth.Grayscale : BitDepth.BlackAndWhite;
            if (args.Length > 2) dpi = int.Parse(args[2]);
            if (args.Length > 3) deviceName = args[3];
        }
        File.AppendAllText(logFile, "output filename: " + filename + Environment.NewLine);
        File.AppendAllText(logFile, "bitDepth: " + bitDepth + Environment.NewLine);
        File.AppendAllText(logFile, "dpi: " + dpi + Environment.NewLine);
        File.AppendAllText(logFile, "deviceName: " + deviceName + Environment.NewLine);

        using (var scanningContext = new ScanningContext(new GdiImageContext()))
        {
            //File.AppendAllText(logFile, "1" + Environment.NewLine);
            var controller = new ScanController(scanningContext);
            //File.AppendAllText(logFile, "2" + Environment.NewLine);
            var scanOption = new ScanOptions
            {
                TwainOptions = new TwainOptions { Dsm = TwainDsm.NewX64, IncludeWiaDevices = true },
                Driver = Driver.Twain,
                Dpi = dpi,
                BitDepth = bitDepth,
                PaperSource = PaperSource.Feeder
            };
            //File.AppendAllText(logFile, "3" + Environment.NewLine);

            ScanDevice selectedDevice = null;
            var devices = await controller.GetDeviceList(scanOption);
            //File.AppendAllText(logFile, "4" + Environment.NewLine);
            foreach (var device in devices)
            {
                if (device.Name == deviceName)
                {
                    selectedDevice = device;
                    break;
                }
            }
            File.AppendAllText(logFile, "devices.Count: " + devices.Count + Environment.NewLine);
            //File.AppendAllText(logFile, "5" + Environment.NewLine);

            if (selectedDevice == null)
            {
                Console.WriteLine("Requested scanner not found");
                return;
            }
            //File.AppendAllText(logFile, "6" + Environment.NewLine);

            scanOption.Device = selectedDevice;

            var images = controller.Scan(scanOption).ToBlockingEnumerable();
            File.AppendAllText(logFile, "images.Count: " + images.Count() + Environment.NewLine);
            var pdfExporter = new PdfExporter(scanningContext);
            //File.AppendAllText(logFile, "7" + Environment.NewLine);
            var procImgs = new Collection<ProcessedImage>();
            //File.AppendAllText(logFile, "8" + Environment.NewLine);
            foreach (var image in images)
            {
                procImgs.Add(image);
            }
            File.AppendAllText(logFile, "procImgs.Count: " + procImgs.Count + Environment.NewLine);
            if (procImgs.Count > 0)
            {
                await pdfExporter.Export(filename, procImgs);
            }
        }
    }
}

public class LadpcScanAppConfiguration
{
    private string _grayInsteadOfBW = "false";

    public string GrayInsteadOfBW
    {
        get { return _grayInsteadOfBW; }
        set { _grayInsteadOfBW = value; }
    }
}

cyanfish, 2023-09-08:
You could try setting ScanningContext.Logger to see if any extra diagnostic information is useful. But if it's working in VS and not otherwise I suspect it's something to do with the command-line arguments and not the NAPS2 part.

shahareldad, 2023-09-10:
I managed to scan via command line though I still don't see any difference in the arguments.
Thanks

closed 2023-09-10 by shahareldad