cyanfish/naps2 #189: Not all scanner are visible in the application

shahareldad, 2023-09-11:
Hi

It's not 100% related to the NAPS2.
I have built a console application and using the NAPS2.SDK and I manage to get all the drivers and scan documents (specifically using Kyocera scanner).
The issue is when I address the console application via Window Service.
The Window Service is exposing an HTTP localhost server to allow users to operate the scanner from the browser.
How ever the scanners list is missing one.
This is what I get:
AV332,AV332IL,TWAIN2 FreeImage Software Scanner
But I'm missing the Kyocera twain drivers.

Do you have any idea where/what to look for?

Thanks for any help

cyanfish, 2023-09-11:
If everything else is the same between console and service, it could be related to the user the service is running as (services run as a system user by default).

Other than that you could try different DSM options and see if that makes a difference. I'm not sure if it works with your scanner but you could also try WIA driver instead of TWAIN.

You could also try changing this setting on the service:
https://www.coretechnologies.com/blog/images/allow-service-to-interact-with-desktop.png

shahareldad, 2023-09-12:
I tried switching to another user (me) instead of the system user, no effect.
I tried using WIA, it showed me 2 drivers where one says the scanner is offline and the other says only glass, no feeder - so no luck here either.
Long ago I already tried the desktop (tried again this morning just in case) - also with no luck.

My next option would be using custom URL but with this option I will be dependent on the browser (chrome for example changes stuff every now and then especially when it comes to running 3rd party apps).

closed 2023-09-13 by shahareldad

rwerg7w230jDF, 2025-02-17:
how did you solve it? I have the exact same issue. Kyocera Twain drivers are visible in the NAPS2 GUI but not in sdk.

shahareldad, 2025-02-20:
I changed from using a windows service to start the scanner to using custom
url protocol (windows registry).
This has more privileges than a windows service thus more drivers are
visible.
I also ditched the NAPS, it was to complicated to switch to, I remained
with using TWAIN.Net library.
https://sourceforge.net/projects/twainnet/