cyanfish/naps2 #193: allow console from portable mode.

BiatuAutMiahn, 2023-09-14:
When attempting to auto scan with the console while using portable mode, it only launches the GUI.

cyanfish, 2023-09-15:
You can use NAPS2.Console.exe which is inside the "App" folder.

BiatuAutMiahn, 2023-09-17:
That's what I'm figured out but profiles aren't portable, or am I overlooking something?

cyanfish, 2023-09-17:
The portable version uses profiles that are stored in the "Data" folder.

BiatuAutMiahn, 2023-09-18:
Right but isn't the naps2.portable wrapper performing redirection? if so, running naps2.console directly  doesn't perform the redirection

cyanfish, 2023-09-18:
No, it's just a launcher. Everything has a separate build for the portable that references the local folders instead of the user folders.

BiatuAutMiahn, 2023-09-18:
Gotcha, thx

closed 2023-09-18 by BiatuAutMiahn