cyanfish/naps2 #196: 7.x.x NAPS2.Worker.exe hangs with fujitsu 7460, 6.1.2 works fine

bani6809, 2023-09-19:
**Describe the bug**
updated to 7.1.0, NAPS hangs on fujitsu 7460
revert to 6.1.2, scans work fine
tried 7.0b1, same problems repro.

**To Reproduce**
Steps to reproduce the behavior:
1. Fujitsu PaperStream IP fi-7460
2. If you have set to "use the native UI", NAPS hangs merely after opening and closing the fujitsu UI.
3. If you have set "Use predefined settings", NAPS hangs after the scan. If you click cancel, only one page of a double sided scan shows up.

**Expected behavior**
It should work like 6.1.2

**Screenshots**
[image]

scanner says scan is finished.

[image]

NAPS never completes, just hangs...

[image]

if you hit cancel, only one side of a double sided scan shows up.

[image]

**Desktop (please complete the following information):**
 - OS: Windows 10
 - Version 7.1.0

**Additional context**
I did get NAPS2.Worker.exe to throw an exception and backtrace at least once, which may give a clue:
```
Status(StatusCode="Unavailable", Detail="failed to connec to all addresses")
Grpc.Core.RpcException: Status(StatusCode="Unavailable", Detail="failed to connect to all addresses")
   at System.Runtime.ExceptionServices.ExceptionDispatchInfo.Throw()
   at System.Runtime.CompilerServices.TaskAwaiter.HandleNonSuccessAndDebuggerNotification(Task task)
   at GrpcDotNetNamedPipes.Internal.MessageReader`1.<MoveNext>d__5.MoveNext()
--- End of stack trace from previous location where exception was thrown ---
   at System.Runtime.ExceptionServices.ExceptionDispatchInfo.Throw()
   at NAPS2.Remoting.Worker.WorkerServiceAdapter.<TwainScan>d__11.MoveNext()
--- End of stack trace from previous location where exception was thrown ---
   at System.Runtime.ExceptionServices.ExceptionDispatchInfo.Throw()
   at System.Runtime.CompilerServices.TaskAwaiter.HandleNonSuccessAndDebuggerNotification(Task task)
   at NAPS2.Scan.Internal.Twain.RemoteTwainSessionController.<StartScan>d__3.MoveNext()
--- End of stack trace from previous location where exception was thrown ---
   at System.Runtime.ExceptionServices.ExceptionDispatchInfo.Throw()
   at System.Runtime.CompilerServices.TaskAwaiter.HandleNonSuccessAndDebuggerNotification(Task task)
   at NAPS2.Scan.Internal.Twain.TwainScanDriver.<Scan>d__3.MoveNext()
--- End of stack trace from previous location where exception was thrown ---
   at System.Runtime.ExceptionServices.ExceptionDispatchInfo.Throw()
   at System.Runtime.CompilerServices.TaskAwaiter.HandleNonSuccessAndDebuggerNotification(Task task)
   at NAPS2.Scan.Internal.RemoteScanController.<Scan>d__5.MoveNext()
--- End of stack trace from previous location where exception was thrown ---
   at System.Runtime.ExceptionServices.ExceptionDispatchInfo.Throw()
   at System.Runtime.CompilerServices.TaskAwaiter.HandleNonSuccessAndDebuggerNotification(Task task)
   at NAPS2.Scan.ScanController.<>c__DisplayClass10_0.<<Scan>b__6>
```

cyanfish, 2023-09-24:
Can you follow the instructions [here](https://www.naps2.com/doc/troubleshooting#debug-log) to collect debug logs?

Dimetr007, 2023-10-23:
Та же проблема с МФУ kyocera.
[debuglog.txt](https://github.com/cyanfish/naps2/files/13068334/debuglog.txt)

closed 2023-12-25 by cyanfish