cyanfish/naps2 #198: NAPS2 7.1.0 will not save PDF (clicking button does nothing, no error)

hrunting, 2023-09-24:
**Describe the bug**
With NAPS2 on Windows 11 x64, I cannot save a PDF of a single-page scan. Clicking on the button Save PDF button in both the main interface and the document preview window results in nothing happening (no error, no save dialog). The functionality works fine on NAPS2 6.1.2

**To Reproduce**
Steps to reproduce the behavior:
1. Scan a document
2. Click on the Save PDF button

**Expected behavior**
A Save Document window appears asking where I should save the PDF and under what filename, and then the app would save the file.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Desktop (please complete the following information):**
 - OS: Windows 11 x64
 - Version 7.1.0

**Additional context**
This happened after an upgrade from NAPS2 6.1.2 to NAPS2 7.1.0.

hrunting, 2023-09-24:
From the error log:

2023-09-24 10:55:49.0321 14744 An error occurred that caused the application to close. System.ArgumentException: The path is not of a legal form.
   at string System.IO.Path.NormalizePath(string path, bool fullCheck, int maxPathLength, bool expandShortPaths)
   at string System.IO.Path.InternalGetDirectoryName(string path)
   at void Eto.WinForms.Forms.WindowsFileDialog<TControl, TWidget>.set_FileName(string value)
   at bool NAPS2.EtoForms.EtoDialogHelper.PromptToSavePdf(string defaultPath, out string savePath)
   at async Task<bool> NAPS2.ImportExport.ExportController.SavePdf(ICollection<UiImage> uiImages, ISaveNotify notify)
   at async Task NAPS2.EtoForms.Desktop.DesktopController.SavePdf()
   at async NAPS2.EtoForms.ActionCommand(Action action)+(?) => { }
   at void System.Runtime.ExceptionServices.ExceptionDispatchInfo.Throw()

The path in PDF settings is fine (C:\Users\<username>\Downloads\), but I tried changing it and that also does not print a dialog box. The error in the error log for that behavior is:

2023-09-24 10:56:49.0873 14744 An error occurred that caused the application to close. System.ArgumentException: The path is not of a legal form.
   at string System.IO.Path.NormalizePath(string path, bool fullCheck, int maxPathLength, bool expandShortPaths)
   at string System.IO.Path.InternalGetDirectoryName(string path)
   at void Eto.WinForms.Forms.WindowsFileDialog<TControl, TWidget>.set_FileName(string value)
   at bool NAPS2.EtoForms.EtoDialogHelper.PromptToSavePdf(string defaultPath, out string savePath)
   at void NAPS2.EtoForms.Widgets.FilePathWithPlaceholders.OpenPathDialog(object sender, EventArgs e)
   at void Eto.PropertyStore.TriggerEvent<T>(object key, object sender, T args)
   at void Eto.Forms.Button.OnClick(EventArgs e)
   at void Eto.Forms.Button+Callback.OnClick(Button widget, EventArgs e)
   at void Eto.WinForms.Forms.Controls.ButtonHandler<TControl, TWidget, TCallback>.Control_Click(object sender, EventArgs e)
   at void System.Windows.Forms.Control.OnClick(EventArgs e)
   at void System.Windows.Forms.Button.OnMouseUp(MouseEventArgs mevent)
   at void System.Windows.Forms.Control.WmMouseUp(ref Message m, MouseButtons button, int clicks)
   at void System.Windows.Forms.Control.WndProc(ref Message m)
   at void System.Windows.Forms.ButtonBase.WndProc(ref Message m)
   at void System.Windows.Forms.Button.WndProc(ref Message m)
   at IntPtr System.Windows.Forms.NativeWindow.Callback(IntPtr hWnd, int msg, IntPtr wparam, IntPtr lparam)

hrunting, 2023-09-24:
I've fixed this by manually editing %APPDATA%\Roaming\NAPS2\config.xml and changing the PdfSettings.DefaultFileName value from a directory to a PDF filename. With an actual file as the value rather than a directory, the save dialogs now appear properly.

If the app detects that a directory is the default (carried over from previous install in my case), it should do something sane (like open a save dialog in that directory, which was the previous behavior).

cyanfish, 2023-09-24:
Do you know what the original value was that was causing the problem? I can't reproduce the error (putting "C:\Users\Ben\Downloads" in PDF Settings doesn't work properly but it doesn't give that error). e.g. are there unicode characters in your username?

hrunting, 2023-09-24:
It was "C:\Users\hrunting\Downloads\" (without the quotes). Trailing backslash, all ASCII characters.

cyanfish, 2023-09-24:
Ah yeah the trailing backslash does it, thanks.

cyanfish, 2023-10-11:
This should be fixed in [7.1.1](https://www.naps2.com/download).

closed 2023-10-11 by cyanfish