cyanfish/naps2 #205: Exception after scan on macOS

frkd-dev, 2023-10-08:
**Describe the bug**
After a page scan, the app throws an exception on macOS. The app is basically unusable anymore.

**To Reproduce**
Steps to reproduce the behavior:
1. Press |> button to start scan
2. Wait until the scan finishes
3. Exception occurred once the scan is completed

**Expected behavior**
No exception

**Screenshots**
[image]

**Desktop (please complete the following information):**
 - OS: macOS 14
 - NAPS2 7.1.0

**Additional context**
Backtraces:
```
System.Exception: System.Exception: Could not initialize an instance of the type 'CoreGraphics.CGImage': handle is null.
It is possible to ignore this condition by setting ObjCRuntime.Class.ThrowOnInitFailure to false.
   at ObjCRuntime.DisposableObject.InitializeHandle(NativeHandle handle, Boolean verify)
   at ObjCRuntime.DisposableObject..ctor(NativeHandle handle, Boolean owns, Boolean verify)
   at CoreFoundation.NativeObject..ctor(NativeHandle handle, Boolean owns, Boolean verify)
   at CoreFoundation.NativeObject..ctor(NativeHandle handle, Boolean owns)
   at CoreGraphics.CGImage..ctor(Int32 width, Int32 height, Int32 bitsPerComponent, Int32 bitsPerPixel, Int32 bytesPerRow, CGColorSpace colorSpace, CGBitmapFlags bitmapFlags, CGDataProvider provider, NFloat[] decode, Boolean shouldInterpolate, CGColorRenderingIntent intent)
   at NAPS2.Scan.Internal.Apple.DeviceOperator.FlushImageWithColorSpace(MemoryStream fullBuffer, ICScannerBandData data)
   at NAPS2.Scan.Internal.Apple.DeviceOperator.<>c__DisplayClass26_0.<DidScanToBandData>b__0()
   at System.Threading.ExecutionContext.RunFromThreadPoolDispatchLoop(Thread threadPoolThread, ExecutionContext executionContext, ContextCallback callback, Object state)
--- End of stack trace from previous location ---
   at System.Threading.ExecutionContext.RunFromThreadPoolDispatchLoop(Thread threadPoolThread, ExecutionContext executionContext, ContextCallback callback, Object state)
   at System.Threading.Tasks.Task.ExecuteWithThreadLocal(Task& currentTaskSlot, Thread threadPoolThread)
--- End of stack trace from previous location ---
   at NAPS2.Scan.Internal.Apple.DeviceOperator.Scan()
   at NAPS2.Scan.Internal.Apple.AppleScanDriver.Scan(ScanOptions options, CancellationToken cancelToken, IScanEvents scanEvents, Action`1 callback)
   at NAPS2.Scan.Internal.RemoteScanController.Scan(ScanOptions options, CancellationToken cancelToken, IScanEvents scanEvents, Action`2 callback)
   at NAPS2.Remoting.Worker.WorkerServiceImpl.Scan(ScanRequest request, IServerStreamWriter`1 responseStream, ServerCallContext context)
   at void NAPS2.Remoting.RemotingHelper.HandleErrors(Error error)
   at async Task NAPS2.Remoting.Worker.WorkerServiceAdapter.Scan(ScanningContext scanningContext, ScanOptions options, CancellationToken cancelToken, IScanEvents scanEvents, Action<ProcessedImage, string> imageCallback)
   at async Task NAPS2.Scan.Internal.WorkerScanBridge.Scan(ScanOptions options, CancellationToken cancelToken, IScanEvents scanEvents, Action<ProcessedImage, PostProcessingContext> callback)
   at async IAsyncEnumerable<ProcessedImage> NAPS2.Scan.ScanController.Scan(ScanOptions options, CancellationToken cancelToken)+(?) => { }
```

williamgeurts, 2023-10-10:
Same problem here with scanning from ADF. Scanning from flatbed works fine. I am on a Apple Silicon Mac.

cyanfish, 2023-10-11:
Can you please:
1. Update to [7.1.1](https://www.naps2.com/download)
2. Follow the instructions [here](https://www.naps2.com/doc/troubleshooting#debug-log) to collect debug logs

frkd-dev, 2023-10-11:
Here are logs acquired from 7.1.1

```
2023-10-11 09:33:33.7981 1412 ICC: Opening session 
2023-10-11 09:33:35.7356 1412 DidOpenSession (null) 
2023-10-11 09:33:35.7412 1412 ICC: Waiting for ready 
2023-10-11 09:33:35.7512 1412 DidBecomeReady 
2023-10-11 09:33:35.7514 1412 ICC: Selecting unit 
2023-10-11 09:33:35.7796 1412 DidSelectFunctionalUnit ICScannerFunctionalUnitFlatbed (null) 
2023-10-11 09:33:35.7883 1412 DidSelectFunctionalUnit ICScannerFunctionalUnitFlatbed (null) 
2023-10-11 09:33:35.7883 1412 ICC: Setting scan parameters 
2023-10-11 09:33:35.7960 1412 ICC: Requesting scan 
2023-10-11 09:33:35.9050 1412 DidReceiveStatusInformation ICScannerStatusWarmingUp 
2023-10-11 09:33:35.9085 1412 DidReceiveStatusInformation ICScannerStatusWarmUpDone 
2023-10-11 09:34:08.8888 1412 DidScanToBandData buffer complete 
2023-10-11 09:34:08.8931 1412 Image data: width 4960, height 7016, type Rgb, comp 4, bits/comp 8, bits/pixel 32, bytes/row 19840, data len 139197440 
2023-10-11 09:34:08.8931 1412 Flushing image with color sync profile /System/Library/Frameworks/ICADevices.framework/Versions/A/Resources/Scanner RGB Profile.icc 
2023-10-11 09:34:08.9073 1412 DidCompleteScan (null) 
2023-10-11 09:34:08.9073 1412 ICC: Waiting for scan results 
2023-10-11 09:34:08.9086 1412 ICC: Closing session (in finally) 
2023-10-11 09:34:09.0274 1412 DidReceiveStatusInformation (null) 
2023-10-11 09:34:09.0274 1412 DidReceiveStatusInformation (null) 
2023-10-11 09:34:09.0312 1129 System.Exception: Could not initialize an instance of the type 'CoreGraphics.CGImage': handle is null.
It is possible to ignore this condition by setting ObjCRuntime.Class.ThrowOnInitFailure to false. System.Exception: System.Exception: Could not initialize an instance of the type 'CoreGraphics.CGImage': handle is null.
It is possible to ignore this condition by setting ObjCRuntime.Class.ThrowOnInitFailure to false.
   at ObjCRuntime.DisposableObject.InitializeHandle(NativeHandle handle, Boolean verify)
   at ObjCRuntime.DisposableObject..ctor(NativeHandle handle, Boolean owns, Boolean verify)
   at CoreFoundation.NativeObject..ctor(NativeHandle handle, Boolean owns, Boolean verify)
   at CoreFoundation.NativeObject..ctor(NativeHandle handle, Boolean owns)
   at CoreGraphics.CGImage..ctor(Int32 width, Int32 height, Int32 bitsPerComponent, Int32 bitsPerPixel, Int32 bytesPerRow, CGColorSpace colorSpace, CGBitmapFlags bitmapFlags, CGDataProvider provider, NFloat[] decode, Boolean shouldInterpolate, CGColorRenderingIntent intent)
   at NAPS2.Scan.Internal.Apple.DeviceOperator.FlushImageWithColorSpace(MemoryStream fullBuffer, ICScannerBandData data)
   at NAPS2.Scan.Internal.Apple.DeviceOperator.<>c__DisplayClass26_0.<DidScanToBandData>b__0()
   at System.Threading.ExecutionContext.RunFromThreadPoolDispatchLoop(Thread threadPoolThread, ExecutionContext executionContext, ContextCallback callback, Object state)
--- End of stack trace from previous location ---
   at System.Threading.ExecutionContext.RunFromThreadPoolDispatchLoop(Thread threadPoolThread, ExecutionContext executionContext, ContextCallback callback, Object state)
   at System.Threading.Tasks.Task.ExecuteWithThreadLocal(Task& currentTaskSlot, Thread threadPoolThread)
--- End of stack trace from previous location ---
   at NAPS2.Scan.Internal.Apple.DeviceOperator.Scan()
   at NAPS2.Scan.Internal.Apple.AppleScanDriver.Scan(ScanOptions options, CancellationToken cancelToken, IScanEvents scanEvents, Action`1 callback)
   at NAPS2.Scan.Internal.RemoteScanController.Scan(ScanOptions options, CancellationToken cancelToken, IScanEvents scanEvents, Action`2 callback)
   at NAPS2.Remoting.Worker.WorkerServiceImpl.Scan(ScanRequest request, IServerStreamWriter`1 responseStream, ServerCallContext context)
   at void NAPS2.Remoting.RemotingHelper.HandleErrors(Error error)
   at async Task NAPS2.Remoting.Worker.WorkerServiceAdapter.Scan(ScanningContext scanningContext, ScanOptions options, CancellationToken cancelToken, IScanEvents scanEvents, Action<ProcessedImage, string> imageCallback)
   at async Task NAPS2.Scan.Internal.WorkerScanBridge.Scan(ScanOptions options, CancellationToken cancelToken, IScanEvents scanEvents, Action<ProcessedImage, PostProcessingContext> callback)
   at async IAsyncEnumerable<ProcessedImage> NAPS2.Scan.ScanController.Scan(ScanOptions options, CancellationToken cancelToken)+(?) => { }
```

cyanfish, 2023-10-12:
Okay, I can confirm this is an issue after upgrading to macOS 14 Sonoma. I will see if I can find a fix.

cyanfish, 2023-10-20:
This should be fixed in [7.1.2](https://www.naps2.com/download).

closed 2023-10-20 by cyanfish

frkd-dev, 2023-10-22:
Thanks a lot! It's fixed now! 🎉