cyanfish/naps2 #206: Bring NAPS2 to Flathub.org

MG-LSJ, 2023-10-13:
Though NAPS2 is already available in flatpak format, it would be great if you release it on flathub.org, as its the default source for many linux distro app stores.

ivdimitro, 2023-11-19:
In theory the submission to flathub can be made from other not strictly the app writer/owner [process](https://docs.flathub.org/docs/for-app-authors/submission/)
@cyanfish will this be something you are OK with or publish to flathub is in your plans anyway?

cyanfish, 2023-11-20:
Yes, I do plan to eventually publish on Flathub (along with the Windows and Mac stores too).

RokeJulianLockhart, 2024-03-22:
> https://github.com/cyanfish/naps2/issues/206#issuecomment-1818023079

https://github.com/microsoft/winget-pkgs/issues/90081#issuecomment-2015496123

rambo919, 2024-04-08:
It's on Flathub now but it has driver permission issues, this can be remedied with something like Flatseal but I don't know which option to use.

repires, 2024-04-23:
> It's on Flathub now but it has driver permission issues, this can be remedied with something like Flatseal but I don't know which option to use.

Where is the link? I didn´t find it. @cyanfish any updates on the publishing? I created a thread on the forum.

https://discourse.flathub.org/t/naps2-not-another-pdf-scanner/6493/2

Hoeze, 2024-12-25:
any updates on this? I would like to have automatic updates for my flatpaks

repires, 2025-04-09:
Do you a third party to add your flatpak?
@cyanfish

muelli, 2025-08-01:
The manifest seems to be [here](https://github.com/cyanfish/naps2/commit/dcb1e045d1fd94783fff8e4193abc678d506baf6#diff-3c922985e077775c2b7641859869a01e2eb0b83d1e9b478d01d0d693fcf1a5e4) in case somebody wants to propose it for Flathub.

repires, 2025-08-10:
Hi @cyanfish . The windows store was added. Are you able to publish on the flathub repo too?

ostfriese4, 2026-10-05:
I'm tryining to update the manifest to use checker-data and newer dependencies. https://github.com/ostfriese4/naps2/blob/flatpak-update/NAPS2.Setup/config/linux/com.naps2.Naps2.yml But the manifest references a nuget-sources.json that I can't find. Does anyone know, where I can find it?

Openhuman2, 2026-10-05:
Seems like something is shown already in Bazaar. 
( no pictures yet and it doesn't seem to be found in search)
I have installed the flatpak manually..
Thanks to whoever did this.

[image]
[image]

ostfriese4, 2026-10-05:
This is because Bazaar does support manually installed flatpaks. So it does only display that you installed it.