cyanfish/naps2 #207: General settings

cyanfish, 2023-10-14:
Elevate some existing config-file-only settings:
- [SaveButtonDefaultAction](https://github.com/cyanfish/naps2/blob/58a0405449b788237d43dae84e291d4cc3af6e71/NAPS2.Lib/Config/CommonConfig.cs#L42)
- [HiddenButtons](https://github.com/cyanfish/naps2/blob/58a0405449b788237d43dae84e291d4cc3af6e71/NAPS2.Lib/Config/CommonConfig.cs#L45C27-L45C40)
- [AlwaysRememberDevice](https://github.com/cyanfish/naps2/blob/58a0405449b788237d43dae84e291d4cc3af6e71/NAPS2.Lib/Config/CommonConfig.cs#L60C17-L60C37)
- [DeleteAfterSaving](https://github.com/cyanfish/naps2/blob/58a0405449b788237d43dae84e291d4cc3af6e71/NAPS2.Lib/Config/CommonConfig.cs#L87C17-L87C34)
- [DisableExitConfirmation](https://github.com/cyanfish/naps2/blob/58a0405449b788237d43dae84e291d4cc3af6e71/NAPS2.Lib/Config/CommonConfig.cs#L93C17-L93C40)
- [SingleInstance](https://github.com/cyanfish/naps2/blob/58a0405449b788237d43dae84e291d4cc3af6e71/NAPS2.Lib/Config/CommonConfig.cs#L96C17-L96C31)
- [ShowPageNumbers](https://github.com/cyanfish/naps2/blob/58a0405449b788237d43dae84e291d4cc3af6e71/NAPS2.Lib/Config/CommonConfig.cs#L171C17-L171C32)

As well as providing access to existing settings (PDF settings, EnableDebugLogging, etc.)

And some potential new settings:
- Disable delete confirmation
- Set default profile on clicking Scan
- ...

muescha, 2025-01-22:
Combine all settings into a the general Settings Dialog (with Tabs?):
- File / PDF-Setting
- File / Image-Settings
- File / Email-Settings

cyanfish, 2025-02-02:
Keyboard shortcuts added in [8.0b3](https://www.naps2.com/download).

cyanfish, 2026-10-25:
In [8.6.0](https://www.naps2.com/download), settings windows are now combined, and you can also remove buttons from the toolbar.