cyanfish/naps2 #214: Error Save PDF with ✅Skip save prompt to non-existent folder

chromotion, 2023-10-24:
**Describe the bug**
NAPS don't create a non-existent folder by clicking *Save PDF* with checked *Skip save prompt*

**To Reproduce**
Steps to reproduce the behavior:
1. Go to PDF Settings
2. Set Default File Path to *C:\Users\username...\Desktop\$(YYYY)-$(MM)-$(DD)\$(hh)$(mm)$(ss).pdf* (folder shouldn't exist)
3. Check mark *Skip save prompt*
4. Scan or Import image
5. Click *Save PDF*
6. See error

**Expected behavior**
NAPS create not existed folder specified in Default File Path by clicking Save PDF

**Desktop (please complete the following information):**
 - OS: Windows 10 x64
 - Version 7.0b1-7.1.2

**Additional context**
v.6.1.2 works fine with same configuration

willibrandon, 2023-10-25:
System.IO.DirectoryNotFoundException: Could not find a part of the path

```
at Microsoft.Win32.SafeHandles.SafeFileHandle.CreateFile(String fullPath, FileMode mode, FileAccess access, FileShare share, FileOptions options)
   at Microsoft.Win32.SafeHandles.SafeFileHandle.Open(String fullPath, FileMode mode, FileAccess access, FileShare share, FileOptions options, Int64 preallocationSize)
   at System.IO.Strategies.OSFileStreamStrategy..ctor(String path, FileMode mode, FileAccess access, FileShare share, FileOptions options, Int64 preallocationSize)
   at System.IO.Strategies.FileStreamHelpers.ChooseStrategyCore(String path, FileMode mode, FileAccess access, FileShare share, FileOptions options, Int64 preallocationSize)
   at System.IO.Strategies.FileStreamHelpers.ChooseStrategy(FileStream fileStream, String path, FileMode mode, FileAccess access, FileShare share, Int32 bufferSize, FileOptions options, Int64 preallocationSize)
   at System.IO.FileStream..ctor(String path, FileMode mode)
   at NAPS2.Pdf.PdfExporter.MergePassthroughPages(MemoryStream stream, String path, List`1 passthroughPages, PdfExportParams exportParams, ProgressHandler progress) in D:\code\naps2\NAPS2.Sdk\Pdf\PdfExporter.cs:line 144
```

chromotion, 2023-10-26:
It's odd that [Save Images] command works propertly, unlike [Save PDF] command.

cyanfish, 2023-12-13:
This should be fixed in [7.2.0](https://www.naps2.com/download).

closed 2023-12-13 by cyanfish