cyanfish/naps2 #218: NixOS Build: Corrupt JPEG data: found marker 0xd9 instead of RST1

batonac, 2023-10-31:
**Describe the bug**
I'm working on packaging NAPS2 for NixOS, and running into an issue that I suspect is a bug. The interface works, my scanners are detected, but when I try to scan I get a series of errors which seem to indicate that there's a mishap somewhere between SANE and NAPS2 in handling image data.

The actual error from the interface is as follows:
```
System.Exception: System.TypeInitializationException: The type initializer for 'GLib.Object' threw an exception.
   at GLib.Object..ctor(IntPtr raw)
   at Gdk.Pixbuf..ctor(Colorspace colorspace, Boolean has_alpha, Int32 bits_per_sample, Int32 width, Int32 height)
   at NAPS2.Images.Gtk.GtkImageContext.Create(Int32 width, Int32 height, ImagePixelFormat pixelFormat) in /build/source/NAPS2.Images.Gtk/GtkImageContext.cs:line 67
   at NAPS2.Scan.Internal.Sane.SaneScanDriver.ProcessSingleFrameImage(SaneReadParameters p, Byte[] data) in /build/source/NAPS2.Sdk/Scan/Internal/Sane/SaneScanDriver.cs:line 281
   at NAPS2.Scan.Internal.Sane.SaneScanDriver.ScanPage(ISaneDevice device, IScanEvents scanEvents, OptionData optionData) in /build/source/NAPS2.Sdk/Scan/Internal/Sane/SaneScanDriver.cs:line 264
   at NAPS2.Scan.Internal.Sane.SaneScanDriver.<>c__DisplayClass7_0.<Scan>b__0() in /build/source/NAPS2.Sdk/Scan/Internal/Sane/SaneScanDriver.cs:line 86
   at System.Threading.Tasks.Task.InnerInvoke()
   at System.Threading.Tasks.Task.<>c.<.cctor>b__272_0(Object )
   at System.Threading.ExecutionContext.RunFromThreadPoolDispatchLoop(Thread , ExecutionContext , ContextCallback , Object )
--- End of stack trace from previous location ---
   at System.Threading.ExecutionContext.RunFromThreadPoolDispatchLoop(Thread , ExecutionContext , ContextCallback , Object )
   at System.Threading.Tasks.Task.ExecuteWithThreadLocal(Task& , Thread )
--- End of stack trace from previous location ---
   at NAPS2.Scan.Internal.RemoteScanController.Scan(ScanOptions options, CancellationToken cancelToken, IScanEvents scanEvents, Action`2 callback) in /build/source/NAPS2.Sdk/Scan/Internal/RemoteScanController.cs:line 40
   at NAPS2.Remoting.Worker.WorkerServiceImpl.Scan(ScanRequest request, IServerStreamWriter`1 responseStream, ServerCallContext context) in /build/source/NAPS2.Sdk/Remoting/Worker/WorkerServiceImpl.cs:line 151
   at void NAPS2.Remoting.RemotingHelper.HandleErrors(Error error) in /build/source/NAPS2.Sdk/Remoting/RemotingHelper.cs:line 22
   at async Task NAPS2.Remoting.Worker.WorkerServiceAdapter.Scan(ScanningContext scanningContext, ScanOptions options, CancellationToken cancelToken, IScanEvents scanEvents, Action<ProcessedImage, string> imageCallback) in /build/source/NAPS2.Sdk/Remoting/Worker/WorkerServiceAdapter.cs:line 83
   at async Task NAPS2.Scan.Internal.WorkerScanBridge.Scan(ScanOptions options, CancellationToken cancelToken, IScanEvents scanEvents, Action<ProcessedImage, PostProcessingContext> callback) in /build/source/NAPS2.Sdk/Scan/Internal/WorkerScanBridge.cs:line 38
   at async IAsyncEnumerable<ProcessedImage> NAPS2.Scan.ScanController.Scan(ScanOptions options, CancellationToken cancelToken)+(?) => { } in /build/source/NAPS2.Sdk/Scan/ScanController.cs:line 149
```
The error printed on the console is simply:
```
Corrupt JPEG data: found marker 0xd9 instead of RST1
```
I suspect that this error is coming from an underlying library, presumably libjpeg. Though it seems from a perusal of the source code that GTK is being utilized for image manipulation, and not libjpeg directly.

My further suspicion is of a version incompatibility somewhere between an upstream library shipped in NixOS unstable and the current NAPS2 code. I can reproduce this exact issue whether building from source or building from the latest Debian package. FWIW, the SANE backend on Ubuntu 22.04 is libsane 1.1.1, on NixOS its 1.2.1.

**To Reproduce**
Steps to reproduce the behavior:
1. Download the [build spec in this gist](https://gist.github.com/batonac/a21d12af1462c688d2a7b56ee089c09b), build with `nix-build`, and run with `./result/bin/naps2`
2. Set up a scanner and scan a page.
3. See error

**Expected behavior**
The image(s) are loaded from the scanner into NAPS2.

**Screenshots**
Here you can see the error in the interface as well as the console:
[image]

**Desktop (please complete the following information):**
 - OS: NixOS unstable
 - Version 7.1.1

**Additional context**
I'm new at building packages for Nix, and haven't even used NAPS2 on Linux up to this point. Initially, I assumed that it was my own fault in the packaging that this isn't working, but the further I went along the more I suspected that there's a library incompatibility somewhere. [Here's the package spec for running the Debian package on Nix](https://gist.github.com/batonac/82467d268a1a8669cd3afacc614afbaa) (I spent much less time perfecting that one).

I'm glad to try to pitch in to further diagnose and/or resolve this issue, but I'm at a loss as to how to proceed, so any further suggestions are much appreciated!

cyanfish, 2023-10-31:
Does using the command-line `scanimage` work for scanning?

batonac, 2023-10-31:
Great question! `scanimage -d 'epsonds:libusb:002:009' --format=jpeg -o scan.jpg` scans and saves a file successfully, but does output the error `Corrupt JPEG data: found marker 0xd9 instead of RST1` to the terminal. I guess you zeroed in on the source of that error.

What should I test next?

mkfrey, 2024-02-18:
This is likely a upstream issue with SANE: https://gitlab.com/sane-project/backends/-/issues/99
I'm also having this issue with other distros. Should be fixed with the next release.

batonac, 2024-02-19:
@mkfrey that's good to know. I haven't spun up a Ubuntu machine yet, to confirm this myself. I'm too hooked on NixOS 😀

cyanfish, 2024-02-21:
@mkfrey Are you having the same TypeInitializerException on other distros?

@batonac One other thing to try on nixOS, is does importing an image work? If not it might be having trouble loading [one of these](https://github.com/GtkSharp/GtkSharp/blob/96f30862e9eea764ea1119c34d401b1c71e7ac57/Source/Libs/Shared/GLibrary.cs#L23) libraries. Maybe there's a missing runtime dependency.

closed 2025-06-08 by cyanfish