cyanfish/naps2 #219: Scan issue causing black border to be added around the scanned image

jahvari, 2023-11-01:
**Describe the bug**
I believe to have an issue with the cropping of a scan. When a smaller sheet (smaller than a regular piece of paper) is scanned, a black border is put around the scan, rather than white space. The issue is when the image is printed, the black is still around the image causing much more black ink to be used. 

**Expected behavior**
In prior versions, white space would be around the image rather than the black border.

**Screenshots**
I will add screenshots shortly.

**Desktop (please complete the following information):**
 - Windows 10
 - Version 7.1.2

cyanfish, 2023-11-06:
Can you add those screenshots when you get a chance? What driver (twain/wia) are you using? What scanner model? Can you confirm that it's different between version 6.1.2 (or whatever prior version) and 7.1.2?

jahvari, 2023-11-06:
@cyanfish Attached are the screenshots from 7.1.2

[image]

[image]

[image]

[image]

jahvari, 2023-11-06:
Sorry, did not mean to close this issue. Attached are screenshots using TWAIN driver vs WIA in 7.1.2.

[image]

[image]

Issue seems to be with the WIA driver and fixed when using TWAIN driver.

cyanfish, 2023-11-10:
What scanner model? Can you confirm that it's different between version 6.1.2 (or whatever prior version) and 7.1.2?

jahvari, 2023-11-15:
The scanner model is: Kodak S2085f

I have confirmed that behavior is present in previous versions (I used v6.1.2). The issue seems to be related when using the WIA driver and having the "Use Predefined settings" button. 

I tested the WIA driver and "Use Native UI" with both v 7.1.2 and v6.1.2 and it cropped correctly.

cyanfish, 2023-11-18:
Okay, to fix this would require supporting "Automatic" page size, which has been on the todo list for a while. I've created #226 to track it.

closed 2023-11-18 by cyanfish