cyanfish/naps2 #226: Automatic page size

cyanfish, 2023-11-18:
- [WIA_PAGE_AUTO](https://learn.microsoft.com/en-us/windows/win32/wia/-wia-wia2-pagesizeconstants)
- [ICAP_AUTOSIZE](https://twain.org/wp-content/uploads/2015/05/TWAIN-2.3-Specification.pdf) (TWAIN 2.0+ only)
- Custom image processing?

cap4cho, 2024-02-08:
I find this option very interesting since the black border affects how the software processes blank pages, any sheet other than A4 does not take the thresholds into account. Is this implemented? Is it something that can be launched from the command line?

doingso, 2024-08-21:
Is there any ongoing work on this feature? I really need an automatic sizing function, but it's not available, which is causing difficulties for me. If there are any updates planned, I would like to hear about it.

justlurking, 2025-06-17:
Not intending to comment spam but just wanted to let you know that autodetecting scan size is my #1 feature request by far. Hope you're able to implement it at some point.

christianclavet, 2025-09-21:
Hi, Have you tried to use the Native Interface and select those features? I have them on most Kodak and Futjitsu scanners (adf)? If your scanner doesnt support this on the native interface, it will not work event if NAPS implement theses:

@[WIA_PAGE_AUTO](https://learn.microsoft.com/en-us/windows/win32/wia/-wia-wia2-pagesizeconstants)
[ICAP_AUTOSIZE](https://twain.org/wp-content/uploads/2015/05/TWAIN-2.3-Specification.pdf) (TWAIN 2.0+ only)

There will be a need to C# custom image processing function to crop the image, and this might be complex as it relate to computer vision.

If you are not using the native interface you loose a lot of functionnalities...

justlurking, 2025-09-22:
> Hi, Have you tried to use the Native Interface and select those features? I have them on most Kodak and Futjitsu scanners (adf)? If your scanner doesnt support this on the native interface, it will [...]

Hi, I appreciate the reply. Could you clarify what you mean by native interface? I like NAPS2 because the scanning software included with my Canon ImageClass 733 all-in-one printer/scanner is atrocious and the build-in image capture software on Mac is very bare bones. I'm on macOS (15.6.1), and I currently seem to be using an Airprint driver for printing (I'm not sure if that also applies to scanning).

I looked at the readme for the latest Canon scanner drivers and it says the latest drivers DO NOT support TWAIN and that I should downgrade to a far older driver if I need TWAIN support. Presumably these latest drivers support ICA, Apple's native scanning interface, though that isn't clear either. 

Based on the links in the original issue and that you quoted above, does NAPS2 depend on the user installing a TWAIN driver in order to function properly on Mac? By the way, the ICAP_AUTOSIZE link is broken for me, but I found the 2.3 specification on Github at https://github.com/twain/twain-specification/tree/master/versions/2.3. Also, version 2.4 appears to be the latest TWAIN spec.

cyanfish, 2026-10-12:
I've added an auto-crop feature in [8.5.0](https://www.naps2.com/download). Either click the button in the Crop window, or check "Auto-crop scanned pages" in your profile settings under Advanced. Please let me know how it works for you, and if it fails on anything, please provide a sample scan (pre-crop) if possible.

axmmisaka, 2026-10-13:
Thanks for this awesome open source tool; I am using Epson DS70 and if I specify a large target size and insert something smaller it will look like this:

[image]

i.e. what's missing horizontally is black, what's missing vertically is white

It seems that auto-crop is not handing this case as expected as it only cuts the white section out, leaving the black section in.

[image]

I'm not an expert but I know some C#, I can take a look at the implementation and see if I could contribute later, thanks

cyanfish, 2026-10-13:
@axmmisaka I pushed some code with a potential fix, feel free to try this [test version](https://1drv.ms/u/c/2918c6d8dc4b6479/IQDlzISpSh-gToM8a_TiXxzVAa89esb-jTeKelGueI9o0X0?e=152XVV).

cyanfish, 2026-10-25:
Published those improvements in [8.6.0](https://www.naps2.com/download).