cyanfish/naps2 #240: No duplex scans

Mrothyr, 2023-11-25:
**Describe the bug**
No duplex scanning. Only the upper side is scanned.

**Expected behavior**
The backside should be scanned.

**Desktop (please complete the following information):**
 - OS: Windows 11 Pro
 - Version 7.1.2.0 x64
 - Epson DS-1660W

**Additional context**
errorlog.txt:
2023-11-25 23:18:20.8020 24600 Error setting property 3088 System.ArgumentException: Der Wert liegt außerhalb des erwarteten Bereichs.
   at void NAPS2.Wia.Native.IWiaPropertyStorage.WriteMultiple(uint cpspec, PROPSPEC[] rgpspec, PROPVARIANT[] rgpropvar, uint propidNameFirst)
   at uint NAPS2.Wia.Native.NativeWiaMethods.SetPropertyInt(IntPtr propStoragePtr, int propId, int value)
   at void NAPS2.Wia.WiaProperty.set_Value(object value)
   at void NAPS2.Wia.WiaExtensions.SetProperty(WiaItemBase item, int propId, int value)
   at void NAPS2.Scan.Internal.Wia.WiaScanDriver+WiaScanContext.SafeSetProperty(WiaItemBase item, int propId, int value)
   
   (first line german for "The value is outside the expected range")

cyanfish, 2023-11-27:
Please try the [troubleshooting](https://www.naps2.com/doc/troubleshooting) steps, in particular try using TWAIN Driver instead of WIA Driver in your profile settings.

Mrothyr, 2023-11-30:
A twain driver is not available.