cyanfish/naps2 #242: ocr imported pdf

lollita, 2023-11-27:
Do it is possible to ocr a pdf imported instead that scanned?

lollita, 2023-11-28:
When I try it not work: import, ocr, save.

cyanfish, 2023-11-28:
Can you attach a sample pdf with the problem?

kintaro1981, 2024-01-08:
same problem here with NAPS 7.22 on Windows 10 (22H2)

to reproduce:
import a scanned PDF
Check if OCR options are selected
Save as PDF

The pdf is not searchable.

example:
https://ft19.comune.lucca.it/f.php?h=2eE7jd6D&d=1

cyanfish, 2024-01-08:
@kintaro1981 You can see the sideways text on the side is selectable/searchable, I assume that's confusing the OCR engine. #258 tracks potential improvements to this sort of thing.

kintaro1981, 2024-01-09:
yes I think that is the same issue.
I hope this will be solved sorting out #258