cyanfish/naps2 #243: ESCL client is sending plain HTTP request to HTTPS endpoint

dantmnf, 2023-11-28:
**Describe the bug**
ESCL client is issuing requests to endpoint like `http://10.10.10.10:443/eSCL/ScannerCapabilities`

**To Reproduce**
Steps to reproduce the behavior:
1. Configure an ESCL scanner profile
2. Click Scan
3. See error

**Expected behavior**
It should scan the document.

**Screenshots**
Get an error window instead.

Messages in debuglog:

```
2023-11-28 16:01:52.1679 25532 Discovered ESCL Service: HP LaserJet MFP M72625dn, ipv4 10.10.10.10, ipv6 (null), port 443, uuid 4969251a-e618-41ab-a046-896b6602ec2a 
2023-11-28 16:01:52.1679 25532 ESCL GET http://10.10.10.10:443/eSCL/ScannerCapabilities 
2023-11-28 16:01:52.1679 25532 Discovered ESCL Service: HP LaserJet MFP M72625dn, ipv4 10.10.10.10, ipv6 (null), port 443, uuid 4969251a-e618-41ab-a046-896b6602ec2a 
2023-11-28 16:01:52.1789 25532 An error occurred while sending the request. System.Net.Http.HttpRequestException: An error occurred while sending the request. ---> System.Net.WebException: 基础连接已经关闭: 连接被意外关闭。
   at WebResponse System.Net.HttpWebRequest.EndGetResponse(IAsyncResult asyncResult)
   at void System.Net.Http.HttpClientHandler.GetResponseCallback(IAsyncResult ar)
   --- 内部异常堆栈跟踪的结尾 ---
   at async Task<HttpResponseMessage> System.Net.Http.HttpClient.FinishSendAsyncUnbuffered(Task<HttpResponseMessage> sendTask, HttpRequestMessage request, CancellationTokenSource cts, bool disposeCts)
   at async Task<string> System.Net.Http.HttpClient.GetStringAsyncCore(Task<HttpResponseMessage> getTask)
   at async Task<XDocument> NAPS2.Escl.Client.EsclClient.DoRequest(string endpoint)
   at async Task<EsclCapabilities> NAPS2.Escl.Client.EsclClient.GetCapabilities()
   at async Task NAPS2.Scan.Internal.Escl.EsclScanDriver.Scan(ScanOptions options, CancellationToken cancelToken, IScanEvents scanEvents, Action<IMemoryImage> callback)
   at async Task NAPS2.Scan.Internal.RemoteScanController.Scan(ScanOptions options, CancellationToken cancelToken, IScanEvents scanEvents, Action<ProcessedImage, PostProcessingContext> callback)
   at async IAsyncEnumerable<ProcessedImage> NAPS2.Scan.ScanController.Scan(ScanOptions options, CancellationToken cancelToken)+(?) => { }

```

**Desktop (please complete the following information):**
 - OS: Windows 10
 - Version 7.1.0

**Additional context**
Add any other context about the problem here.

cyanfish, 2023-11-29:
Can you try this one and attach the debuglog (from the Data folder in the extracted zip)?

[naps2-7.1.2-test3.zip](https://github.com/cyanfish/naps2/files/13495730/naps2-7.1.2-test3.zip)

dantmnf, 2023-11-29:
This build looks (some sort of) good, but seems something in my scanner went wrong:
[debuglog - Copy.txt](https://github.com/cyanfish/naps2/files/13496102/debuglog.-.Copy.txt)

dantmnf, 2023-11-29:
Nevermind it became operational after a few attempts.

cyanfish, 2023-11-29:
Thanks, I think I need to add some retry capability for retrieving the scanned documents, that should hopefully make it more reliable.

cyanfish, 2023-12-13:
[7.2.0](https://www.naps2.com/download) should be more reliable when connecting.

closed 2023-12-13 by cyanfish

cyanfish, 2023-12-17:
And [7.2.1](﻿﻿https://www.naps2.com/download) should fully support HTTPS connections.