cyanfish/naps2 #244: ID card scan mode - combine two scans into one

davoper, 2023-12-01:
Automatically combine two scans of both sides of ID card or other similar images and combine them into a singe scan showing both sides one below the other.

If one desires to show both sides of ID card in the same image additional image processing and manual combining is required.
Similar feature already exists in scan/copy machines (called 2-sided ID scanning or similar)

This could work with a regular flat scanner where you would have two scans of front and back sides and then for instance it you select two scanned images and press function to combine them into one the resulting image would merge useful parts of each original images into new image and place them next to each other.

Savasdotexe, 2024-01-21:
I like this and agree it would be quite useful. I guess by combine them into a single scan, you mean combine them into a single page. I think rather than below, side by side or an option whether it's landscape or portrait.

So if it was to fit in an A4, maybe it can scan with A5 for both then merge it together? I have no idea how this would be done, but I think adding this as a profile template would be the easiest? Just throwing some ideas out there.

uderbashi, 2024-02-19:
Quick report for someone who wants to look into this sometime in the future.

I wanted to see how tough this feature is to implement quickly, and I was thinking it would be as simple as combining the rotate and resize transforms.

I found out that these transforms (found in NAPS2.Images/Transforms/) seem to be carried out for each image individually.
I could not find a very straightforward way to implement this quickly with transforms. 
Myabe some manipulation of PdfPage (NAPS2.Sdk/Pdf/Pdfium/)?

cyanfish, 2024-03-11:
This is available in [7.4.0](https://www.naps2.com/download) (look for "Combine" in the Image menu).

closed 2024-03-11 by cyanfish

sweenu, 2024-04-02:
For an ID card, I would want the front and back to appear on one A4 page, not on two A4 pages stitched to one another like it'd done with Combine.

cyanfish, 2024-04-02:
@sweenu You may need to crop before combining.

uderbashi, 2024-04-02:
@sweenu also consider making the original scans in A5 size for the dimensions to look better. Since A5+A5=A4