cyanfish/naps2 #249: mac mini m2 pro with osx 14.2 -- the page size is not working

cvraosan, 2023-12-14:
In mac mini m2 pro with OSX 14.1.2 -- the page size is not working

what ever size you select it scans image to a4 size only

i checked the latest version 7.2 and 7.1 

This issue started just now

Selected us legal paper size but it scans only a4 size

cyanfish, 2023-12-14:
What scanner model? Are you using "Apple Driver" in your profile settings? Is the scanner connected by USB or over the network?

cvraosan, 2023-12-15:
HiUsing HP 7740Connected over ethernet It was working nicely Even now a4 scanning is ok all the other features are working But paper size change does nit work It always scans at a4 sizeSent from 📱

cvraosan, 2023-12-15:
Yes and apple driver selectedSent from 📱

cvraosan, 2023-12-15:
HiAnd another thing is now i have a vpnExpress vpn which is turned onI did not try to test with it turned offWill do that and report back to youSent from 📱

cyanfish, 2023-12-15:
A couple things to try:
1. Can you change the page size if you use the built-in [macOS scanning](https://support.apple.com/en-ca/guide/mac-help/mh28032/mac)?
2. Does it work if you select "ESCL Driver" in your NAPS2 profile settings?

cvraosan, 2023-12-15:
HiYes i tried with builtin apple scanning solution and it works perfectly for all types of page sizesESCL does not work in your programRgdsSent from 📱

A couple things to try:

cvraosan, 2023-12-15:
Hi
Please find attached the setup screens of the NAPS2 on my Mac mini m2 pro

￼￼￼

Please see the full scanned image got from scanning using the MAC Scanning builtin software 
￼

cvraosan, 2023-12-15:
hi

I downgraded to 7.09b-mac-univ.pkg 
and tried  here is the error
￼
￼

UnciaPrima, 2023-12-15:
Hello,

I'm new to SNAPS2 but i think i'm experiencing the same problem (or a very similar one). I'm trying to scan A4 pages.
- Setting to greyscale generates a very weird image, very tilted and wrongly cropped. 
- Setting to Color scale generates a normal PDF file.

Scanning with Apple's Image Capture is perfectly normal.

Brother ADS-1200 / Macbook Air 2013 / macOS 11.7.3

cyanfish, 2023-12-17:
The grayscale issue should be fixed in [NAPS2 7.2.1](https://www.naps2.com/download). I've also made a change that could help with page size but I'm not sure.

Please:
1. Try and see if page size works in 7.2.1 with Apple Driver.
2. If not, try again with ESCL Driver. If that doesn't work, please let me know what error you get etc. (p.s. I can't see images you send in email attachments, you'll have to reply on Github if you want to upload images)

cvraosan, 2023-12-17:
Hi Ben
the bug is fixed -- now working well in 7.2.1

cvraosan, 2023-12-17:
closed working well -- very good program and fast response from developer

closed 2023-12-17 by cvraosan

UnciaPrima, 2023-12-17:
> The grayscale issue should be fixed in [NAPS2 7.2.1](https://www.naps2.com/download). I've also made a change that could help with page size but I'm not sure.
>
> Please:
>
>     1. Try and see if page [...]

Thank you very much for your quick answer and fix! I will be using NAPS2 from now!