cyanfish/naps2 #250: [i18n] Certain string translated in .po files but not in .resx files

Matthaiks, 2023-12-17:
**Describe the bug**
The `Shared scanners can be used from other computers on the local network by selecting "ESCL Driver" in the other computer's NAPS2 profile settings.` string displays always in English because it has been translated in .po files only (https://github.com/cyanfish/naps2/commit/b737ceba4f7631354a00599af2fd8f7e445468cb). 

**To Reproduce**
Steps to reproduce the behavior:
1. Switch to Czech, Italian, Hungarian or Polish
2. Go to Scanner Sharing
3. See the string in English

**Expected behavior**
The string should be translated.

**Screenshots**
[image]

**Desktop (please complete the following information):**
 - OS:  Windows 10 22H2
 - Version: 7.2.1

Matthaiks, 2023-12-22:
Fixed in 7.2,2. Thanks!

closed 2023-12-22 by Matthaiks

isaak654, 2023-12-23:
> Fixed in 7.2,2. Thanks!

Not really, just switch to the Czech or Italian language:
[image]