cyanfish/naps2 #253: NAPS2 Background sharing

DestructiveSilence, 2023-12-17:
Is it possible to make NAPS2 automatically start as a Windows service, and share scanners in the background?

cyanfish, 2023-12-17:
I wondered how long it would take someone to ask :)

This is definitely planned but not currently possible yet (at least not without using the [SDK](https://github.com/cyanfish/naps2/blob/master/NAPS2.Sdk.Samples/NetworkSharingSample.cs)).

wildwind123, 2025-04-11:
Maybe it would be better to make background service to separate program ? For better development and maintenance ? They can exist independently.

wildwind123, 2025-04-11:
For me make tray will enough, where I describe problem there
https://github.com/cyanfish/naps2/issues/596

cyanfish, 2025-06-08:
This has been added in [8.2.0](https://www.naps2.com/download). If you check the box in the Scanner Sharing window it will start on login (but it does require login, it's not a system-level service).

closed 2025-06-08 by cyanfish