cyanfish/naps2 #254: Range of IP addresses for eSCL driver.

DestructiveSilence, 2023-12-18:
Could you add the ability to specify a range of IP addresses to search for devices via the eSCL driver?

Right now it only displays devices located on the same subnet as the computer (for me it is 10.12.**69**.xx). 
But beyond that, we have many MFPs in other ranges (like 10.12.**60**.xx, 10.12.**64**.xx, and so on).

DestructiveSilence, 2023-12-18:
[image]
[image]

cyanfish, 2023-12-18:
ESCL uses mDNS for service discovery, which involves both NAPS2 and the scanners sending multicast packets to UDP 224.0.0.251:5353 (which get delivered to everything on the network).

If it's not finding the scanners, that means these packets aren't getting delivered across the subnet boundary. It should in theory be possible to configure the router between subnets to forward the packets, but how to do that depends on the hardware.

cyanfish, 2023-12-18:
It's not an enterprise environment, but an example: https://blog.christophersmart.com/2020/03/30/resolving-mdns-across-vlans-with-avahi-on-openwrt/

You might also look for the term "Bonjour Gateway" in relation to whatever networking hardware you have.

cyanfish, 2024-02-05:
Tracking documentation updates in #302

closed 2024-02-05 by cyanfish