cyanfish/naps2 #256: Usability issue: There is no "OCR" button to let the user launch the OCR function manually (when the "auto OCR" is disabled)

raindropsfromsky, 2023-12-20:
**Describe the bug**
The "OCR" button in the toolbar is meant to set OCR-related configurations of NAPS2.
It cannot launch the OCR function manually.

If the user has disabled the "auto OCR", he does not have a button to manually launch the OCR function.

Also, the "Automatically run OCR after scanning" option explicitly links the OCR function with scanning.

But what if the user has not done any scanning at all? 
- What if he has imported the image/pdf file? 
- What if he has dragged-and-dropped the file into NAPS2?

If the user has disabled the auto-OCR option, he does not expect NAPS2 to carry out OCR automatically even for imported or drag-n-dropped items also. So he looks for a manual method to OCR the document/image. He sees a large button marked "OCR", but that only offers him configurations, without actually launching the OCR operation.

Thus, the user has no clue what to do next!

Secondly, the toolbar mixes up buttons belonging to different functional groups, as shown below:
[image]
Note that the user cannot use these groups in the "left-to-right" order sequence.

**To Reproduce**
As described above

**Expected behavior**

All the configuration settings should be kept separately.
This includes settings for OCR, language selection, etc.
The most widely accepted convention is to provide a "Preferences" button at the right end of toolbar. 
Typically, it has a "gear" or "wrench+screwdriver" icon.
A common dialog can cover all settings-related options as tabs or divs.

The "Profiles" button is settings-related, not part of the input process.
Therefore, it should be hived off to the "Preferences" dialog.

All the inputs-related buttons should be kept together in the left compartment of the toolbar.
Thus, the "Scan" and "Import" buttons should be placed side-by-side.

The OCR button should be split in two parts:
- The configuration-related options should be hived off to the "Preferences" section.
- A new "OCR" button should be added in the second compartment of the toolbar. <br> The only function of this button is to manually launch the OCR function. <br> It should not provide any options for program settings.

Thus, the toolbar would be split in four streamlined compartments, as shown below:
[image]1. Inputs (Scan, import)
2. OCR (in manual mode)
3. Export (save as pdf, save as images, email, print)   
4. Preferences (program options) (set profile, set OCR mode, etc.)

The user would be able to work linearly across the toolbar, from left-to-right, as shown below:
[image]

**Screenshots**
Provided inline.

**Desktop (please complete the following information):**
 Not relevant: This is a UI design issue.
 
**Additional context**

cyanfish, 2023-12-20:
I do agree that "Automatically run OCR after scanning" could use a note of clarification, something along the lines of "This can speed up PDF saving later. If not selected OCR will be run while the PDF is being saved." But I don't consider a manual "Run OCR" function a high priority since it's only an optimization. (At some point I might add a function to see/copy the OCR text which would in fact need a way to manually run OCR, but that's not a high priority either.)

I strongly disagree that core functionality like OCR and Profiles should be hidden behind a settings dialog. I do intend to add a settings dialog soon-ish, but only for non-core customization. I prioritize discoverability and easy access over strict consistency.

Savasdotexe, 2024-01-24:
I actually came to a similar conclusion and thought to suggest the ability to customize the bars, just in case there isn't a perfect, one fit solution for everyone - https://github.com/cyanfish/naps2/issues/283

My suggestion https://github.com/cyanfish/naps2/issues/282 to have the OCR highlighted, may solve this confusion. Thinking further, you could replace the dialog box and make the OCR button be dropdown with language selection and other options to enable.

cyanfish, 2024-02-06:
I've created #306 to track toolbar change suggestions.

Tarek-Hasan, 2024-02-06:
> I strongly disagree that core functionality like OCR and Profiles should be hidden behind a settings dialog.

@cyanfish I don't mind profiles button, but next to that is the scan's down arrow button, which reveals a new profile option. It doesn't add any value other than visual clutter. You should remove this repetitive option.

cyanfish, 2024-02-06:
You can't edit or delete profiles from the menu.

Tarek-Hasan, 2024-02-06:
I was talking about removing "New profile" menu, not the "Profile" button. It shouldn't exist as everything profile related can be done by profile button next to it.

cyanfish, 2024-04-08:
I've renamed the toggle in [NAPS2 7.4.1](https://www.naps2.com/download) from "Automatically run OCR after scanning" to "Pre-emptively run OCR after scanning" which hopefully communicates the meaning better.

closed 2024-04-08 by cyanfish

kaffeeringe, 2024-10-25:
Thank you for this idea. I found this issue while looking for information if NAPS2 supported OCR on existing PDFs. It's not intuitive.