cyanfish/naps2 #258: Please allow user to define blocks manually and re-assign their types (text, image, table)

raindropsfromsky, 2023-12-21:
**Is your feature request related to a problem? Please describe.**
NAPS2 decides the type of content automatically, and OCRs the "text" and "table" type of blocks, while reproduces the "image" type of blocks without OCR. However, there is a problem: Some images contain embedded text. Such images are also treated as text blocks. The result is suboptimal.

**Describe the solution you'd like**

Let NAPS2 show the default block types to the user, and allow user to edit those blocks.
- If he draws a block that covers more than one blocks, they should be merged into a new block.
- Let him also re-assign the type (image/text/table) to each block.

All professional OCR software have this feature.

**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.

**Additional context**
Add any other context or screenshots about the feature request here.

kintaro1981, 2024-01-09:
@raindropsfromsky 
I reported the same problem in #242 
in the meanwhile have you solved with some alternatives?

EmperorArthur, 2024-01-22:
While not the exact same issue, related to #208.