cyanfish/naps2 #263: Support HighDPI in Windows

arnowelzel, 2023-12-24:
**Is your feature request related to a problem? Please describe.**
It seems, the Windows version is not DPI aware and when using it with scaling on a 4K display, the UI does not look very nice.

**Describe the solution you'd like**
It would be nice if the application had DPI awareness enabled and if needed using icons and bitmap rendering with upscaling to keep the original size.

**Describe alternatives you've considered**
I tried to enable advanced system upscaling in the application properties in Windows, but this did not help.

**Additional context**
See the screenshot here - this demonstrates the issue. The title bar is in rendered at the native resolution while the application content is blurred as it is only upscaled due to the missing DPI awareness:

[image]

Another example - the profile edit dialog:

[image]

arnowelzel, 2023-12-24:
I started adding an `app.manifest` file to the application with the respective settings and use the need initialization. So getting DPI awareness is quite simple. However there is still work to do to handle the stored Window size as DPI independent value and to handle toolbar icons and bitmaps as device independent images which upscale if needed. Also the dialog element sizes may need to be adjusted depending on the display scaling:

[image]

However text and native Windows Forms controls in dialog windows already look much nicer this way:

[image]

cyanfish, 2023-12-24:
#182 is a previous issue for this. My current plan is to do this after #81 which I think will make addressing element sizes easier.

closed 2023-12-24 by cyanfish