cyanfish/naps2 #264: Feature Request: Copy to Clipboard as JPG

dan-willett, 2023-12-27:
When copying via Control-C in Windows, the format copied is a PNG which results in a too large file for many applications. I would like to see a setting where you can set the format that is copied to the clipboard as JPG instead of PNG so that I can paste a smaller file when copying and pasting.  Currently to get a jpg I have to save it first and then load the file.

This generates an otherwise un-needed file and makes it more difficult ... example use case, copy from NAPS2 and paste into Trello -- if the clipboard image was JPG I would be able to do this.

cyanfish, 2023-12-27:
That's not really how the clipboard works, it doesn't get copied as PNG or any particular format, it just gets copied as an image. How it gets interpreted is up to the receiving app.

dan-willett, 2023-12-31:
That may be true, however it is also true that the format pasted can be controlled by the program making the data available to the clickboard.   For example, if I copy an image using the printscreen key, it pastes into the app as a .BMP, if I copy an image using the Windows clipping tool, it pastes into the program as a .JPG and when I copy an image from NAPS2, it pastes into the program as a .PNG ... now if the clipboard was only using a generic image format, why are the results different depending on what program creates the copy?

dan-willett, 2023-12-31:
Here is some additional information on the subject:  https://stackoverflow.com/questions/6561417/how-do-you-copy-paste-image-data

Also, when I asked Chat GPT 4 the following question:
In a windows program written in C++ how to I put image data on the windows clipboard so that when it is pasted into another app it pastes in jpg format?
Among other things and code samples, it stated:
Set the Clipboard Data:

Use SetClipboardData to place the data onto the clipboard. Since you're dealing with an image in JPG format, you should use the CF_DIB format, which stands for "Device Independent Bitmap". It's a bit tricky because you're actually putting a DIB format on the clipboard, but the data in it is a JPG. Applications that paste from the clipboard will generally accept DIB and interpret it correctly.

cyanfish, 2024-01-03:
Well, the GPT answer is wrong, but apparently there are [undocumented](https://bugzilla.mozilla.org/show_bug.cgi?id=1717306) JPEG (JFIF) and PNG clipboard formats. Can you try this test version and see if it works?
[naps2-7.2.2-testjpegclipboard.zip](https://github.com/cyanfish/naps2/files/13823647/naps2-7.2.2-testjpegclipboard.zip)

cyanfish, 2024-01-15:
I've added the additional formats in [NAPS2 7.3.0](https://www.naps2.com/download). If you get a chance to try please let me know if that helped.

closed 2024-01-15 by cyanfish

dan-willett, 2024-01-17:
I tried it out, but I couldn't find the setting to set it to use JPG for the clipboard...  Am I missing something?