cyanfish/naps2 #268: Error after scan in Twain mode

virtualos, 2023-12-30:
Hello,

OS -> Windows 10 x64 22h2
Naps2 -> v7.2.2 Portable & MSI
Scanner -> HP scanjet 5590

Error:
I made a profile with Twain and WIA driver. WIA driver scans ok, Twain scans ok too but with Twain I got an error at the end:

`2023-12-30 13:45:51.6868 6772 Status(StatusCode="Unavailable", Detail="failed to connect to all addresses") Grpc.Core.RpcException: Status(StatusCode="Unavailable", Detail="failed to connect to all addresses")
   at async Task<bool> GrpcDotNetNamedPipes.Internal.MessageReader<TMessage>.MoveNext(CancellationToken cancellationToken)
   at async Task NAPS2.Remoting.Worker.WorkerServiceAdapter.TwainScan(ScanOptions options, CancellationToken cancelToken, ITwainEvents twainEvents)
   at async Task NAPS2.Scan.Internal.Twain.RemoteTwainSessionController.StartScan(ScanOptions options, ITwainEvents twainEvents, CancellationToken cancelToken)
   at async Task NAPS2.Scan.Internal.Twain.TwainScanDriver.Scan(ScanOptions options, CancellationToken cancelToken, IScanEvents scanEvents, Action<IMemoryImage> callback)+(?) => { }
   at async Task NAPS2.Scan.Internal.RemoteScanController.Scan(ScanOptions options, CancellationToken cancelToken, IScanEvents scanEvents, Action<ProcessedImage, PostProcessingContext> callback)
   at async IAsyncEnumerable<ProcessedImage> NAPS2.Scan.ScanController.Scan(ScanOptions options, CancellationToken cancelToken)+(?) => { }
`

Thanks and waiting for you inspection.

virtualos, 2023-12-30:
I just went to Twain profile -> Advanced -> Twain Implementation -> Select "OLD DSM"
Error solved :)

closed 2023-12-30 by cyanfish

therentabrain, 2024-07-17:
I'm getting this same error and I can't choose a scanner at all as a result.  The Old DSM option didn't help.  Thoughts?