cyanfish/naps2 #272: Development issues regarding the use of scanning in escl driver

zhangxihuai, 2024-01-10:
Hello.
I am currently working on .net6 to build the program, I can use "ScanController. GetDeviceList()" to obtain wia devices in Windows and sane devices in Linux. But I cannot retrieve the device from the "ScanController. GetDeviceList (Driver. Escl)" method, whether it is Windows or Linux. 
Using in NAPS2 7.2.2, I can obtain devices through the "ESCL Driver"，but my own program cannot do it
What should I do?
Thanks

cyanfish, 2024-01-10:
Try setting ScanningContext.Logger (e.g. [NLog](https://www.nuget.org/packages/NLog.Extensions.Logging), or just [implement](https://github.com/cyanfish/naps2/blob/c785204f30a922a73799e511481cb3bd0418d1c7/NAPS2.Sdk.Tests/ContextualTests.cs#L168) a basic ILogger with Console.WriteLine or something). ESCL should have some useful debug-level logging.

I'm not sure what the problem would be, maybe something to do with firewall access from your process?

cyanfish, 2024-01-15:
Thanks, it looks like there's an error that's not logged properly, can you try the latest SDK version (0.2.3) and collect the same logs again?

cyanfish, 2024-01-31:
Closing for now, feel free to comment again if you need more help.

closed 2024-01-31 by cyanfish