cyanfish/naps2 #275: Illegible notifications

LinuxOnTheDesktop, 2024-01-13:
**Describe the bug**
The program gives almost totally illegible notifications. Once I took a screenshot and magnified the notifications, I could read them - and I discovered that one of the notifications was missing some word(s).

**To Reproduce**
1. Run the program, when some notification will be shown.
2. See some notification(s).
3. Try to read the notification(s).

**Expected behavior**
Legibility. No words missing.

**Screenshots**
[image]

**Desktop (please complete the following information):**
 - OS: Windows 10 Pro - 22H2 (2009: 19045). Theme settings:
[image]
 
 - Version [of naps2, you mean]: 7.1.2.0

**Additional context**
I see from the Github releases page that version 7.2.2 fixes some bugs. Yet, the lack of a change log (and me not being a git wizard) prevents me from seeing whether the problem that I report is fixed by that version. (I installed the update, but having done so there were no notifications showing.)

cyanfish, 2024-01-13:
This looks like a duplicate of #225 and should be fixed in 7.2.2

closed 2024-01-13 by cyanfish

LinuxOnTheDesktop, 2024-01-13:
> should be fixed in 7.2.2[.]

Good stuff. Does the fix include a fix for the missing word, though? Also: wouldn't it be a good idea to provide changelogs?

cyanfish, 2024-01-14:
I should have a fix for the missing word in the next version (coming in the next few days). I do have changelogs for new features etc. but tracking every single bug fix is too much overhead for me.