cyanfish/naps2 #276: saving scan as TIFF file fails with standard (k)ubuntu 23.10 - libtiff version mismatch?

ljwobker, 2024-01-14:
On a stock (as of January 2024) install of (k)ubuntu 23.10, NAPS2 errors out ails to save a scan in TIFF format.

```
lwobker@bmax:/usr$ naps2 console --version
naps2 7.2.2+586f9ed987e16ddaf6d2ccf723cd9c8e8381238e
```

the following command ("platen150" is a known good profile for any other file format):
`naps2 console -v -o /tmp/scan.tiff -f -p platen150`

returns:
```
Beginning scan...
Starting scan 1 of 1...
Scanned page 1.
1 page(s) scanned.
Exporting...
An error occurred when trying to save the file.
Finished saving images to /tmp/scan.tiff
```

Initially, the "Finished saving images..." error is VERY misleading, because no TIFF file actually gets created!

```
lwobker@bmax:/usr$ ls -al /tmp/scan.tiff
ls: cannot access '/tmp/scan.tiff': No such file or directory
```
If we look at the error log, we get a much better suspect, it seems that NAPS2 can't find the TIFF library:

```
lwobker@bmax:/usr$ tail ~/.config/naps2/errorlog.txt 
   at bool NAPS2.ImportExport.Images.SaveImagesOperation.Start(string fileName, Placeholders placeholders, IList<ProcessedImage> images, ImageSettings imageSettings, string overwriteFile, bool batch)+() => { } [0]
2024-01-14 15:40:37.9827 6804 An error occurred when trying to save the file. System.DllNotFoundException: Unable to load shared library 'libtiff.so.5' or one of its dependencies. In order to help diagnose loading problems, consider using a tool like strace. If you're using glibc, consider setting the LD_DEBUG environment variable: 
/usr/lib/naps2/libtiff.so.5: cannot open shared object file: No such file or directory
/usr/lib/naps2/liblibtiff.so.5: cannot open shared object file: No such file or directory
/usr/lib/naps2/libtiff.so.5.so: cannot open shared object file: No such file or directory
/usr/lib/naps2/liblibtiff.so.5.so: cannot open shared object file: No such file or directory

   at IntPtr NAPS2.Images.Gtk.LibTiff.TIFFOpen(string filename, string mode)
   at bool NAPS2.Images.Gtk.LibTiffIo.SaveTiff(IList<IMemoryImage> images, string path, TiffCompressionType compression, ProgressHandler progress)
   at bool NAPS2.ImportExport.Images.SaveImagesOperation.Start(string fileName, Placeholders placeholders, IList<ProcessedImage> images, ImageSettings imageSettings, string overwriteFile, bool batch)+() => { } [0]

```

At first I thought it was just a missing package, but after confirming all of the TIFF libraries appear to be installed correctly(?) -- I think it may be that in 23.10 they stopped including the older libraries by default, and even installing libtiff5 didn't create them?

```
lwobker@bmax:/usr$ sudo find /usr -type f -name \*libtiff\*
/usr/share/man/man3/libtiff.3tiff.gz
/usr/lib/x86_64-linux-gnu/libtiff.a
/usr/lib/x86_64-linux-gnu/pkgconfig/libtiff-4.pc
/usr/lib/x86_64-linux-gnu/libtiffxx.so.6.0.1
/usr/lib/x86_64-linux-gnu/libtiff.so.6.0.1
/usr/lib/x86_64-linux-gnu/libtiffxx.a
```

Going back to our original error log, it appears NAPS2 is hunting for libtiff.so.5, so let's see if we can fake it out with a symlink...

```
root@bmax:/usr/lib/x86_64-linux-gnu# ln -s  libtiff.so libtiff.so.5
root@bmax:/usr/lib/x86_64-linux-gnu# file libtiff.so.5 
libtiff.so.5: symbolic link to libtiff.so
```
Now, if we try again, things work!

```
lwobker@bmax:/usr$ naps2 console -v -o /tmp/scan.tiff -f -p platen150
Beginning scan...
Starting scan 1 of 1...
Scanned page 1.
1 page(s) scanned.
Exporting...
Exporting image 1 of 1...
Exporting image 2 of 1...
Finished saving images to /tmp/scan.tiff
lwobker@bmax:/usr$ file /tmp/scan.tiff 
/tmp/scan.tiff: TIFF image data, little-endian, direntries=13, height=1650, bps=26638, compression=LZW, PhotometricInterpretation=RGB, orientation=upper-left, width=1275
```

I don't know enough about linux libraries/packaging to know if this is a "breakage" in the k/ubuntu side, or whether NAPS2 can be simply updated to accept libtiff6 as an acceptable library, or whether there's something more complex going on.  But at least with my very hack-ish and totally unscalable symlink workaround, the scan does run without error and generates a valid, readable TIFF file.

cyanfish, 2024-01-14:
Thanks, that's very helpful. I'll include a fix in the next version to support libtiff.so.6.

cyanfish, 2024-01-15:
This should be fixed in [NAPS2 7.3.0](https://www.naps2.com/download).

closed 2024-01-15 by cyanfish