cyanfish/naps2 #280: "About" window appearing when pressing F1 though not assigned

chyron8472, 2024-01-19:
**Describe the bug**

After reassigning F1 to a <ScanProfile> and NOT to<About>, About window still appears when pressing F1 unless a Scan Profile is otherwise called.

**To Reproduce**
Steps to reproduce the behavior:
1. Assign <About> in appsettings.xml as any other hotkey besides F1, or leave it blank.
2. Assign <ScanProfile1> as F1.
3. Open NAPS2.
5. Press F1. -- BUG: observe "About" window open

-- Scan Profile 1 does list F1 as its hotkey in the dropdown.
-- If any scan is attempted (by means other than pressing F1), _then_ F1 calls Scan Profile 1 and not About window.

**Expected behavior**
Setting <ScanProfile1>F1</ScanProfile1> and <About>Ctrl+F1</About> in appsettings.xml should **always** call Scan Profile 1 when F1 is pressed.

**Desktop (please complete the following information):**
 - OS: Windows 10
 - Version 7.3.0.0 Portable

psp-oscar, 2024-01-24:
Possible another related issue, the main menu/ribbon items "About" & "Settings" disappeared from Version 7.3.0.0 (7.2.2.0 was ok) in case naps2 interface is set to any(?) language except English. 
OS: xUbuntu 18.04 & 22.04 deb.

cyanfish, 2024-01-30:
@psp-oscar Feel free to create another issue, but you should be able to just make the window a bit wider for it to show up.

cyanfish, 2024-02-05:
This should be fixed in [7.3.1](https://www.naps2.com/download).

closed 2024-02-05 by cyanfish