cyanfish/naps2 #281: page size for SANE is not set

danielk117, 2024-01-20:
**Describe the bug**
I'm not able to scan page height above 279.364mm, but the scanner and the SANE driver support it! Very basic example above 279.364mm is DIN A4, which has a page height of 297mm.

**To Reproduce**
Steps to reproduce the behavior:
1. Set page size to A4
2. Scan an A4 page
3. The scan is missing around 20mm at the end (see screenshot)

**Expected behavior**
The scan contains the full content of the A4 pape.

**Screenshots**
[image]

**Desktop (please complete the following information):**
 - OS: MacOS 13
 - Version 7.3.0

**Additional context**

I've enabled the debug logging in NAPS2 and it seems, that it is limiting the page height values to 279.364mm.

```
2024-01-20 15:32:00.8799 9048 Option: 6 page-width Page width Specifies the width of the media.  Required for automatic centering of sheet-fed scans. Fixed Mm 4 SoftSelect, SoftDetect Range 0-224,8457160296025/0,021164263370717938 
2024-01-20 15:32:00.8808 9048 Option: 7 page-height Page height Specifies the height of the media. Fixed Mm 4 SoftSelect, SoftDetect Range 0-879,4887159533074/0,021164263370717938 
2024-01-20 15:32:00.8808 9048 Option: 8 tl-x Top-left x Top-left x position of scan area. Fixed Mm 4 SoftSelect, SoftDetect Range 0-215,87220569161516/0,021164263370717938 
2024-01-20 15:32:00.8808 9048 Option: 9 tl-y Top-left y Top-left y position of scan area. Fixed Mm 4 SoftSelect, SoftDetect Range 0-279,3640192263676/0,021164263370717938 
2024-01-20 15:32:00.8808 9048 Option: 10 br-x Bottom-right x Bottom-right x position of scan area. Fixed Mm 4 SoftSelect, SoftDetect Range 0-215,87220569161516/0,021164263370717938 
2024-01-20 15:32:00.8808 9048 Option: 11 br-y Bottom-right y Bottom-right y position of scan area. Fixed Mm 4 SoftSelect, SoftDetect Range 0-279,3640192263676/0,021164263370717938
```

When I'm configuring a custom page size of 210x210mm for example, the log shows:

```
2024-01-20 14:55:00.6887 9002 Maybe setting resolution 
2024-01-20 14:55:00.6887 9002 Setting resolution to 300 
2024-01-20 14:55:00.6931 9002 Maybe setting tl-x 
2024-01-20 14:55:00.6931 9002 Setting tl-x to 2,936102845807582 
2024-01-20 14:55:00.6943 9002 Maybe setting tl-y 
2024-01-20 14:55:00.6945 9002 Setting tl-y to 0 
2024-01-20 14:55:00.6945 9002 Maybe setting br-x 
2024-01-20 14:55:00.6945 9002 Setting br-x to 212,93610284580757 
2024-01-20 14:55:00.6956 9002 Maybe setting br-y 
2024-01-20 14:55:00.6956 9002 Setting br-y to 210 
```
Everthing working as expected. I've got a scan with 210x210mm.

When setting page size as A4, the log shows:

```
2024-01-20 14:59:03.6001 9005 Maybe setting resolution 
2024-01-20 14:59:03.6003 9005 Setting resolution to 300 
2024-01-20 14:59:03.6040 9005 Maybe setting tl-x 
2024-01-20 14:59:03.6042 9005 Setting tl-x to 2,936102845807582 
2024-01-20 14:59:03.6042 9005 Maybe setting tl-y 
2024-01-20 14:59:03.6052 9005 Setting tl-y to 0 
2024-01-20 14:59:03.6052 9005 Maybe setting br-x 
2024-01-20 14:59:03.6052 9005 Setting br-x to 212,93610284580757 
2024-01-20 14:59:03.6063 9005 Maybe setting br-y 
2024-01-20 14:59:03.6063 9005 Setting br-y to 279,3640192263676 
```

So I've installed `sane-backend` and tried scanning with `scanimage`.

`scanimage --help` shows me, that I'm able to scan width until 224.846mm and height until 879.489mm. 
The width of 215.872mm and height of 279.364mm are presetted for page size and the max values for -l/-x or -t/-y.

```
  Geometry:
    --page-width 0..224.846mm (in steps of 0.0211639) [215.872]
        Specifies the width of the media.  Required for automatic centering of
        sheet-fed scans.
    --page-height 0..879.489mm (in steps of 0.0211639) [279.364]
        Specifies the height of the media.
    -l 0..215.872mm (in steps of 0.0211639) [0]
        Top-left x position of scan area.
    -t 0..279.364mm (in steps of 0.0211639) [0]
        Top-left y position of scan area.
    -x 0..215.872mm (in steps of 0.0211639) [215.872]
        Width of scan-area.
    -y 0..279.364mm (in steps of 0.0211639) [279.364]
        Height of scan-area.
```

 So when I use `scanimage -d fujitsu:fi-6130dj:4953 --source "ADF Duplex" --page-width 210mm --page-height 297mm --format=jpeg -o scan3.jpg`, it scans the whole A4 page. No missing area at the end.

[image]

So I think (with my very limited knowledge about NAPS2 and SANE), NAPS2 just sets top-left x/y and bottom-right x/y positions and limit these values, as expected. But page width and height is never set.

cyanfish, 2024-01-22:
I'll have a look at that, but an off-topic question - how did you install sane-backends? Was that from homebrew etc or built from the latest source?

danielk117, 2024-01-22:
Yes, I installed it using homebrew. It's not the latest source, but the latest released version 1.2.1.

I've got a hint from the SANE project (see https://gitlab.com/sane-project/backends/-/issues/4769): 

> t,l,x,y are constrained by page-width and page-height. You must tell the scanner how big the paper is, before you tell it how big the image will be

When calling `scanimage --page-width 210mm --page-height 297mm --help`, I get the follow output (and new maximum values for -l/-x and -t/-y):

```
  Geometry:
    --page-width 0..224.846mm (in steps of 0.0211639) [210.01]
        Specifies the width of the media.  Required for automatic centering of
        sheet-fed scans.
    --page-height 0..879.489mm (in steps of 0.0211639) [296.994]
        Specifies the height of the media.
    -l 0..210.01mm (in steps of 0.0211639) [0]
        Top-left x position of scan area.
    -t 0..296.994mm (in steps of 0.0211639) [0]
        Top-left y position of scan area.
    -x 0..210.01mm (in steps of 0.0211639) [210.01]
        Width of scan-area.
    -y 0..296.994mm (in steps of 0.0211639) [296.994]
        Height of scan-area.
```

danielk117, 2024-02-03:
@rwobrecht I don't know which platform you're using, but if it is a Mac too, than I've got workaround.

I've build [AirSane](https://github.com/SimulPiscator/AirSane) and using the "apple driver" in NAPS2. It seems that AirSane also reads the top-left and bottom-right values from SANE and has problems with the larger page size, like NAPS. But in Airsane you're able to set options (see SimulPiscator/AirSane#64) and overriding the defaults (on mac with homebrew the option file is located under `/usr/local/etc/airsane/options.conf`).

cyanfish, 2024-02-05:
This should be fixed in [7.3.1](https://www.naps2.com/download).

I've also created a SANE issue for the root cause: https://gitlab.com/sane-project/backends/-/issues/730

closed 2024-02-05 by cyanfish

rwobrecht, 2024-02-05:
> This should be fixed in 7.3.1.
I can confirm the fix is working in Debian 12.4, kernel 6.1.0-17-amd64 

Thanks!

danielk117, 2024-02-05:
Yes, in 7.3.1 it is working. Thank you for fixing!