cyanfish/naps2 #283: Feature Suggestion: Ability to move toolbars to the bottom, be able to resize and lock toolbars also.

Savasdotexe, 2024-01-21:
**Is your feature request related to a problem? Please describe.**
Not really a problem, more of a visual appearance improvement suggestion/ideas.

**Describe the solution you'd like**
- [ ] Option to enable a row and be able to move a toolbar down to the bottom (I think the save options would be better default at the bottom, for a more user friendly appearance).
- [ ] Separate the move and adjustment functions from the Save and export functions and also from Scan/Profiles/Import options to separate toolbars so user can move them according to preference.
- [ ] Separate the OCR and create a button for Scanner Sharing in this group also.
- [ ] Right click to adjust the size of the toolbar, I guess super large, large, etc. This is mostly for the profile icons being small.
- [ ] Ability to right click and Lock toolbars.
- [ ] Combine the Settings, Language & about to a hamburger menu button.

**Describe alternatives you've considered**
More cosmetic improvements, so nothing exists as an alternative.

**Additional context**
Not needed I think as the descriptions make sense, if not happy to share a crappy paint version.

Savasdotexe, 2024-01-24:
> Is there any practical reason for these or just personal preference? I probably won't do most/all of this as toolbar management across Windows/Mac/Linux is already complicated enough.

I wanted to suggest this to be default, I thought this would simplify and organize the interface, purely for a better user experience.

Thinking it might be a personal preference so I changed my suggestion to be customizable so I totally understand if you disagree or more beneficial improvements take president, this was just a suggestion and maybe later down the line.

cyanfish, 2024-02-06:
I've created #306 to track toolbar change suggestions.

closed 2024-02-06 by cyanfish