cyanfish/naps2 #289: Feature Request: Automated Scanner Source Selection Based on Paper Detection

JPluess, 2024-01-23:
Feature Request: Automated Scanner Source Selection Based on Paper Detection

Description:
I propose an enhancement for NAPS2 where the scanner's source automatically switches based on paper detection. Specifically, if paper is detected in the Automatic Document Feeder (ADF), the scanner should use the ADF. If no paper is detected in the ADF, the scanner should default to the glass source. This feature would streamline the scanning process, reducing manual input and enhancing user convenience.

Current Behavior:
Currently, the user must manually select between the ADF and glass scanner sources. 

Expected Behavior:
With this feature implemented, NAPS2 would first check for paper in the ADF. If paper is present, it uses the ADF. If the ADF is empty, it automatically switches to the glass source. This automatic detection and switching would provide a more efficient and user-friendly scanning experience.

German translation is incomplete, shall i complete it? let me know

MaurizioLoreti, 2024-07-10:
I have an Epson WF 2935 scanner, and the Epson scan app has this feature.  Provided that I like NAPS2 better than Epson software, could you add this feature in the future?  Would be useful, and NAPS2 already detects an empty document feeder.

stephankoenig, 2025-03-31:
I would love to have this. My Brother software supports it. I assume it is in the printer firmware if it gets a general request. It takes the ADF when there is paper; otherwise, it takes the glass.

NINI1988, 2026-02-15:
First of all: amazing project. It’s one of very few projects which runs locally without any issues. Thanks for that.

While looking into the codebase, I noticed that there is already an `Auto `value in `PaperSource`:
https://github.com/cyanfish/naps2/blob/3cd6800f936d425d3949589b9966804a039ebd54/NAPS2.Sdk/Scan/PaperSource.cs#L8-L12

However, this option does not seem to be exposed in the GUI.

Additionally, when using a TWAIN driver and selecting “Feeder”, the behavior already appears to be effectively “auto”: it uses the ADF if paper is present and falls back to glass otherwise.

So I’m trying to understand:

- Is there a specific reason why PaperSource.Auto is not exposed in the UI?

- Is this value intended for internal/backend use only?

- Are there backend inconsistencies (e.g. WIA vs TWAIN vs eSCL) that prevent a generic “Auto” option?

I’m just trying to better understand the design decisions before exploring possible changes.

As a temporary **workaround**, using TWAIN with “Feeder” seems to provide similar behavior.

clemensrieder, 2026-09-30:
Yes, please! Manually selecting the corresponding profile is such a pain. I'd really like this to work out of the box on macOS. On Windows, NAPS2 seems to work perfectly with the TWAIN driver, but running a VM just for NAPS2 is kinda cumbersome.

waylan, 2026-11-16:
Any progress on this? This is a problem for me on the command line. My script can call with `--source feeder` and if it fails, then retry with `--source glass`. However, I am also using `--import` and `--email`. The `import` option on the first call is creating a document from the imported file only (without scanning) and attaching it to an email. I have to reject that email (close it without sending) before the call returns and only then can my script fall back to the second call. It would be so much better if that was all handled internally with an `auto` option.