cyanfish/naps2 #290: Feature Request: Allow user to set default Reorder action.

onnoj, 2024-01-28:
**Is your feature request related to a problem? Please describe.**
I have a scanner with a automatic document feeder, but no automatic duplex scanning.
I only scan documents infrequently (sometimes only once every 3 months), and every time I have to figure out which 'Reorder' option is right for me, by trial-and-error.

**Describe the solution you'd like**
It would be great to be able to set a default option for the 'Reorder' button, and move the current functionality of the reorder button (which shows a menu of Interleave/Deinterleave/etc) to a dropdown button (same as for "Save PDF" and "Save Images"), where clicking the 'reorder' button uses the configured default.

**Describe alternatives you've considered**
Alternatively, it could be helpful to have more descriptive naming (although I don't know what) for the reorder functions, or perhaps a smart wizard where the user selects example pages that are two-sided from the already scanned images and the wizard figures out the rest, or perhaps an 'Undo last action' to make the trial&error less frustrating.

**Additional context**
I'm running with the assumption that the user will always end up with the same option anyway, since it depends on how their document feeder works and/or their workflow. ie, I assume I will always end up with the "alternate interleave" option because of the way my scanner works. 

Example image:
[image]

onnoj, 2024-01-28:
Also, I'm happy that NAPS2 exists ❤️

cyanfish, 2024-02-05:
Tracking "Manual Duplex" rework in #303

closed 2024-02-05 by cyanfish