cyanfish/naps2 #294: failes to find scanner when internet blocked by firewall

TungstenE2, 2024-01-31:
OS: Windows11
Version: 7.3.0.0

Click on Scan button.

Firewall detectes outgoing calls.

Connection to IP of scanner is allowed, all others blocked.

Autoupdate disabled.

Scanner not found when connection to internet is blocked.

Why is the app at all trying to make outbound calls?

[image]

[image]

cyanfish, 2024-01-31:
If you're using the ESCL Driver, it uses mDNS to connect to the scanner, which requires sending packets to the multicast address 224.0.0.251:5353, which essentially asks the local network "what devices are on this network?". It doesn't involve any connection outside the local network.

See https://en.wikipedia.org/wiki/Multicast_DNS

cyanfish, 2024-02-05:
Tracking documentation updates in #302

closed 2024-02-05 by cyanfish