cyanfish/naps2 #295: Look for user-specific appsettings.xml

source124, 2024-01-31:
**The problem**

Changes made in appsettings.xml don't survive app upgrades (portable version).

**The solution**
If I copy appsettings.xml into the Data folder, use that instead of the App folder one.

**Describe alternatives you've considered**
I have to be very careful every time I update the app.

cyanfish, 2024-02-02:
What settings do you change in appsettings.xml? For some of them (e.g. ClearAfterSaving, SaveButtonDefaultAction) I've added a Settings window in 7.3.0 that lets you set it on a user level.

Keyboard shortcuts is maybe the main one that still needs to be added (see #61). But there is an undocumented way to make it permanent - you can copy the <KeyboardShortcuts> element into the user-level config.xml file and it will work the same as if it was in appsettings.xml.

source124, 2024-02-05:
It was many things, shortcuts was one of them (others were page size, scan quality,...).

Thanks for the info. Does this apply to any element or the keyboard shortcuts only?

cyanfish, 2024-02-06:
Not every element but a lot. e.g. DefaultProfileSettings for page size, scan quality, etc. will work.

cyanfish, 2024-04-09:
I'm going to close this in favor of #207 as the plan is to provide a UI to change these settings (and as discussed, config.xml can be used in the meantime).

closed 2024-04-09 by cyanfish