cyanfish/naps2 #297: Fails to import document after scanning Canon MF3010

ostasevych, 2024-02-03:
Hi! I used to use naps2 quite long time ago, converting usual MFAs to the network scanning/printing devices with the help of OpenWrt+p910nd+saned on the small router side (eg nexx or gl.inet), and the sanetwain middleware on the windows clients.

Last week I've got Canon MF3010, which is good with local scanning/printing, however there's an issue when trying to perform networking scanning. The naps2 client initiates the scanning but fails to import the results of the scanning using internal (native or memory) transfer without any error. I tried different options for the transfer, none of them were working. 

Though, using the native sanetwain interface it works well, but it complicates the usage of the software.

Is it possible to fix that somehow? If yes, how?

cyanfish, 2024-02-03:
Please follow [these instructions](https://www.naps2.com/doc/troubleshooting) to collect debug logs.

You can also try setting up [AirSane-OpenWrt](https://github.com/cmangla/AirSane-openwrt) which will let NAPS2 connect using "ESCL Driver" directly, no middleware needed.

ostasevych, 2024-02-08:
Hi!

Here are the logs:
```
2022-09-07 08:32:20.8833 Помилка драйвера сканування. NAPS2.Scan.Wia.Native.WiaException: WIA error code 80210001
   at NAPS2.Scan.Wia.Native.WiaDeviceManager.PromptForDevice(IntPtr parentWindowHandle)
   at NAPS2.Scan.Wia.WiaScanDriver.PromptForDeviceInternal()
   at NAPS2.Scan.ScanDriverBase.PromptForDevice()
2022-12-02 16:48:42.3080 Помилка драйвера сканування. NAPS2.Scan.Wia.Native.WiaException: WIA error code 800706BE
   at NAPS2.Scan.Wia.Native.WiaDeviceManager.PromptForDevice(IntPtr parentWindowHandle)
   at NAPS2.Scan.Wia.WiaScanDriver.PromptForDeviceInternal()
   at NAPS2.Scan.ScanDriverBase.PromptForDevice()
2022-12-06 12:29:35.9014 Помилка драйвера сканування. NAPS2.Scan.Wia.Native.WiaException: WIA error code 80210001
   at NAPS2.Scan.Wia.Native.WiaDeviceManager.PromptForDevice(IntPtr parentWindowHandle)
   at NAPS2.Scan.Wia.WiaScanDriver.PromptForDeviceInternal()
   at NAPS2.Scan.ScanDriverBase.PromptForDevice()
```

Please, have a look there.

I will try to experiment with AirSane, though this will require update of the OpenWrt firmware (there's not so much space on Nexx devices for that, so I am using there quite old version).

ostasevych, 2024-02-10:
Sadly, the device has only 8M so airsane is not possible to install, at least by now :(

Regarding the debug logs, here are they:

```
2024-02-10 16:32:46.3773 11080 Debug: Using new dsm. 
2024-02-10 16:32:46.5100 11080 Debug: NTwain internal message loop is starting. 
2024-02-10 16:32:46.5479 11080 Debug: Thread 14: OpenManager. 
2024-02-10 16:32:46.5957 11080 Debug: Using TWAIN2 memory functions. 
2024-02-10 16:32:46.5977 11080 Debug: Source id = 1 
2024-02-10 16:32:46.5977 11080 Debug: Source id = 2 
2024-02-10 16:32:46.5977 11080 Debug: Source id = 3 
2024-02-10 16:32:46.5977 11080 Debug: Thread 14: CloseManager. 
2024-02-10 16:32:50.8747 10640 Debug: Using new dsm. 
2024-02-10 16:32:50.9595 10640 Using TWAIN DSM: C:\Program Files\NAPS2\lib\_win32\twaindsm.dll 
2024-02-10 16:32:50.9595 10640 NAPS2.TW - Opening session 
2024-02-10 16:32:50.9784 10640 Debug: Thread 1: OpenManager. 
2024-02-10 16:32:51.0233 10640 NAPS2.TW - StateChanged (to 3) 
2024-02-10 16:32:51.0233 10640 Debug: Using TWAIN2 memory functions. 
2024-02-10 16:32:51.0233 10640 NAPS2.TW - Finding source 
2024-02-10 16:32:51.0233 10640 Debug: Source id = 1 
2024-02-10 16:32:51.0233 10640 NAPS2.TW - Opening source 
2024-02-10 16:32:51.0422 10640 NAPS2.TW - Name: SaneTwain; Manu: Herman Kuiper; Family: SaneTwain bridge; Version: 1.37 Version 1.37; Protocol: 1.8 
2024-02-10 16:32:51.0422 10640 Debug: Thread 1: OpenSource. 
2024-02-10 16:32:54.9140 10640 NAPS2.TW - StateChanged (to 4) 
2024-02-10 16:32:54.9140 10640 Debug: Registered callback2 OK. 
2024-02-10 16:32:54.9140 10640 NAPS2.TW - Configuring source 
2024-02-10 16:33:13.1889 10640 NAPS2.TW - Enabling source 
2024-02-10 16:33:13.1960 10640 Debug: Thread 1: EnableSource with NoUI. 
2024-02-10 16:33:14.3702 10640 NAPS2.TW - Finishing with error 
2024-02-10 16:33:14.3702 10640 Debug: Thread 1: CloseSource. 
2024-02-10 16:33:14.3702 10640 Debug: Registered callback2 OK. 
2024-02-10 16:33:14.3792 10640 Debug: Thread 1: CloseManager. 
2024-02-10 16:33:14.6746 10640 NAPS2.TW - StateChanged (to 2) 
```

Please, give your advice, what may be improved?

cyanfish, 2024-02-10:
It looks like NAPS2 isn't logging the error correctly. Can you use this test version, turn on debug logging in it, and try again? (And then the debuglog.txt file will be in the Data folder of the extracted zip.)

[naps2-7.3.1-testtwain.zip](https://github.com/cyanfish/naps2/files/14230619/naps2-7.3.1-testtwain.zip)

ostasevych, 2024-02-13:
> It looks like NAPS2 isn't logging the error correctly. Can you use this test version, turn on debug logging in it, and try again? (And then the debuglog.txt file will be in the Data folder of the [...]

Here is the debug log:

```
2024-02-13 14:05:08.0929 2452 Debug: Using new dsm. 
2024-02-13 14:05:08.1836 2452 Using TWAIN DSM: C:\Users\Administrator\Downloads\naps2-7.3.1-testtwain\App\lib\_win32\twaindsm.dll 
2024-02-13 14:05:08.1946 2452 NAPS2.TW - Opening session 
2024-02-13 14:05:08.2035 2452 Debug: Thread 1: OpenManager. 
2024-02-13 14:05:08.2534 2452 NAPS2.TW - StateChanged (to 3) 
2024-02-13 14:05:08.2534 2452 Debug: Using TWAIN2 memory functions. 
2024-02-13 14:05:08.2534 2452 NAPS2.TW - Finding source 
2024-02-13 14:05:08.2534 2452 Debug: Source id = 1 
2024-02-13 14:05:08.2534 2452 NAPS2.TW - Opening source 
2024-02-13 14:05:08.2723 2452 NAPS2.TW - Name: SaneTwain; Manu: Herman Kuiper; Family: SaneTwain bridge; Version: 1.37 Version 1.37; Protocol: 1.8 
2024-02-13 14:05:08.2723 2452 Debug: Thread 1: OpenSource. 
2024-02-13 14:05:12.7055 2452 NAPS2.TW - StateChanged (to 4) 
2024-02-13 14:05:12.7184 2452 Debug: Registered callback2 OK. 
2024-02-13 14:05:12.7184 2452 NAPS2.TW - Configuring source 
2024-02-13 14:05:26.9707 2452 NAPS2.TW - Enabling source 
2024-02-13 14:05:26.9707 2452 Debug: Thread 1: EnableSource with NoUI. 
2024-02-13 14:05:28.9020 2452 NAPS2.TW - Finishing with error NAPS2.Scan.Exceptions.AlreadyHandledDriverException: Exception of type 'NAPS2.Scan.Exceptions.AlreadyHandledDriverException' was thrown.
   at void NAPS2.Scan.Internal.Twain.TwainSessionScanRunner.Init()
2024-02-13 14:05:29.1295 2452 Debug: Thread 1: CloseSource. 
2024-02-13 14:05:29.1295 2452 Debug: Registered callback2 OK. 
2024-02-13 14:05:29.1295 2452 Debug: Thread 1: CloseManager. 
2024-02-13 14:05:29.2620 2452 NAPS2.TW - StateChanged (to 2) 
```

ostasevych, 2024-02-18:
@cyanfish I am sorry to ask, are any updates on that?

cyanfish, 2024-02-18:
Yes, in this test version I've added a new option "Show native TWAIN progress" under "Advanced" in your profile settings, can you try checking that and see if it makes a difference?

[naps2-7.3.1-testtwain2.zip](https://github.com/cyanfish/naps2/files/14323846/naps2-7.3.1-testtwain2.zip)

ostasevych, 2024-02-24:
Hi @cyanfish! Are any updates on this issue?

cyanfish, 2024-02-24:
Sorry, I don't think there's anything else I can do for this, it's just a SaneTwain bug.

cyanfish, 2024-03-11:
Closing in favor of #327 (since SANEWinDS is open source and thus fixable)

closed 2024-03-11 by cyanfish