cyanfish/naps2 #299: System.ArgumentException: Value does not fall within the expected range with Windows 11

n00ris, 2024-02-04:
**Describe the bug**
I own an Plustek Ephoto z300:

https://plustek.com/us/products/film-photo-scanners/ephoto-z300/

If I try to add in a new profile, I get a value out of range error.
This is on a Windows 11.

It is working with a Windows 10 system.

NAPS2, Version 7.3.0

**To Reproduce**
Steps to reproduce the behavior:
1. Go to 'New Profile'
2. Select WIA driver (scanner vendor does not provide a TWAIN driver, at least I did not find any)
3. Click on 'Choose device'
4. Select the scanner
6. See error
```
System.ArgumentException: Value does not fall within the expected range.
   at NAPS2.Wia.Native.IWiaDevMgr2.SelectDeviceDlg(IntPtr hwndParent, Int32 lDeviceType, Int32 lFlags, String& pbstrDeviceID)
   at NAPS2.Wia.Native.NativeWiaMethods.SelectDevice2(IntPtr deviceManagerPtr, IntPtr hwnd, Int32 deviceType, Int32 flags, String& deviceId, IntPtr& device)
   at NAPS2.Wia.WiaDeviceManager.PromptForDevice(IntPtr parentWindowHandle)
   at NAPS2.EtoForms.WinForms.WinFormsInvoker.<>c__DisplayClass4_0`1.<InvokeGet>b__0()
   at object System.Windows.Forms.Control.MarshaledInvoke(Control caller, Delegate method, object[] args, bool synchronous)
   at object System.Windows.Forms.Control.Invoke(Delegate method, params object[] args)
   at T NAPS2.EtoForms.WinForms.WinFormsInvoker.InvokeGet<T>(Func<T> func)
   at async Task<ScanDevice> NAPS2.Scan.ScanPerformer.PromptForDevice(ScanOptions options) x 2
```

**Expected behavior**
The scanner is selected and you can go ahead with further configuration.

**Desktop (please complete the following information):**
 - OS:  Windows 11

cyanfish, 2024-02-05:
Does it work if you try and use Windows Fax and Scan?

In your profile settings, under Advanced->WIA Version, you can try selecting "1.0".

AndyP2, 2024-02-06:
I get the same problem, Win11 with both WIA and TWAIN drivers. Downgrading to v6.1.2 got it working with the twain driver again, but WIA (v1 and v2) still says "the selected scanner does not support using duplex". 
Not sure if that helps at all? I am using HP 8270

cyanfish, 2024-02-06:
@AndyP2 Please open a [new thread](https://github.com/cyanfish/naps2/discussions/new?category=support) as that's a different completely different scanner/driver.

AndyP2, 2024-02-06:
> @AndyP2 Please open a [new thread](https://github.com/cyanfish/naps2/discussions/new?category=support) as that's a different completely different scanner/driver.

fair enough :-)

n00ris, 2024-02-06:
> Does it work if you try and use Windows Fax and Scan?
> 
> In your profile settings, under Advanced->WIA Version, you can try selecting "1.0".

Hello, 
looks like the option does not matter. Same issue with 1.0 and 2.0.

I did the following:
    Go to 'New Profile'
    Select Advanced
    Select WIA Version: 1.0
    Click OK
    Click on 'Choose device'
    Select the device
    See error